stereolabs/zed-ros-wrapper #4: NO_GPU_COMPATIBLE: GeForce 980ti in Ubuntu 14.04

theDrGray, 2015-10-09:
When ever I follow the method shown on the wiki for installing cuda, the Nvidia driver sends my installation into a login loop. Because of this I had to switch to installing cuda w apt-get. All of 6.5 was installed fine and catkin_make finally worked to build the package. Now when I run the launch, the SDK tells me that my 980 is not compatible even though this exact build works fine on the windows side. 

My exact error is the following:

ZED SDK >> (Init) No compatible device found, disparity map unavailable8 - 0
[ INFO] [1444375194.755167596]: NO_GPU_COMPATIBLE

sl-braun, 2015-10-12:
Hi,

Can you check that you have installed cuda_6.5.19 toolkit. This version has support for GeForce GTX 9XX (970 and 980). It may comes from that point.
The link is here : https://developer.nvidia.com/cuda-downloads-geforce-gtx9xx

Best,
Olivier.

theDrGray, 2015-10-12:
It turns out that I had some cuda 7.5 so files left over from a prior installation and the SDK was using them to compile for some reason. I ran: 

sudo apt-get remove cuda-*
sudo apt-get autoremove 
sudo apt-get install cuda-6.5 

then installed the SDK and everything is working great.

As a side note, were using depth_image_proc the register the depth image then rtabmap_ros to pull down the VO and mapping for anyone who gets here via the googles.

closed 2015-10-13 by adujardin

theDrGray, 2015-10-15:
Turns out this has only been masked, not solved. I have access to the camera via teh wrapper, but instead of using the GPU its threadlocking itself to a single thread on my CPU and maxing out its bandwidth. 

I installed Cuda 6.5 after adding the 6.5.-19 repo, but cuda installed version 6.5.-16 and will not let me change the version with any method Ive tried. Tried using the deb and run files provided in the above link. Tried forcing via a pref file. Have not had any luck. Do you know if this is directly related to the cuda version by any chance?

theDrGray, 2015-10-15:
I believe it is in the wrapper itself. I notice another wrapper for the Zed is using OpenGL instead of OpenCV, as are the GPU test samples. 

I ran the zed samples and the GPU tests all utilize my GPU. The second I use the wrapper, my GPU usage drops and a single thread spikes. 

https://i.imgur.com/CTmG8Oh.jpg

https://i.imgur.com/rASKhGk.jpg

theDrGray, 2015-10-15:
It is most certainly the wrapper as this wrapper utilizes the GPU properly:

https://github.com/Myzhar/ros_zed_cuda_driver

sl-braun, 2015-10-20:
Hi,

The major part of the ZED SDK is running on GPU for sure, since it is using CUDA and there is no way to run CUDA on a CPU. 
the device parameter in the init() function will look for the faster CUDA-capable GPU device, therefore if the function return success, your GTX980 has been taken by the ZED SDK.

A high CPU usage may come from the check of suscribers that returns none. Then no grab is done, therefore no GPU process, and the while loop is running too fast. We update the wrapper with a sleep when there is no suscribers to slow down the while loop.

theDrGray, 2015-11-01:
While I appreciate and believe that is the approach, in practice I am seeing a different result on 4 different machines. 

With the stereolabs wrapper I see spiked CPU and ~10% GPU usage and some dropped frames or lag. With both the stereolabs explorer program and the Myzhar wrapper (simple and regular) I see fluctuating CPU usage and almost maxed out GPU usage with 0 dropped frames or lag. The latter is what I am assuming is the proper result.  

I could have an issue with opencv or Cuda, but I doubt that as it is working with your standalone application and the Myzhar wrapper.