stereolabs/zed-ros-wrapper #6: opencv2

droter, 2015-10-19:
Hi,

I am trying to install the zed ros wrapper and I am not able to install the opencv2.  It looks like ros has moved to opencv3.  I also tried to install the dependancies using rosdep install -i -y --from-paths ~/catkin_ws/src/zed-ros-wrapper but that didn't work.

I downloaded the ZED and CUDA drivers from stereolabs.com and have them installed but having some build problems with opencv2.

Thanks for your help,

Matt

adujardin, 2015-10-20:
Hi, 

You can try compiling and installing OpenCV 2.4 from source, the following modules should be sufficient: core highgui imgproc flann and calib3d.
The wrapper should find the OpenCV libraries installed in the system. 

If you encounter an error regarding the missing ros opencv2 package, try deleting "build_depend>opencv2</build_depend>" and "<run_depend>opencv2</run_depend>" from package.xml.

Let me know if this works

droter, 2015-10-20:
Thanks I was able to download OpenCV2.4 from source and build.  If I run rosversion opencv2 it doesn't find it.  I removed the opencv2 from the package.xml.

closed 2015-10-20 by droter