stereolabs/zed-ros-wrapper #8: Fail after Using ZED Camera

droter, 2015-10-21:
Hi,

I think I am close to getting ZED up and running.  When I run the lauch script I am getting:
process[zed_wrapper_node-2]: started with pid [10885]
[ INFO] [1445443609.787001051]: ZED_WRAPPER Node initialized
[ INFO] [1445443609.796179599]: Using ZED Camera
[zed_wrapper_node-2] process has died [pid 10885, exit code -11, cmd /home/matt/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node 0 __name:=zed_wrapper_node __log:=/home/matt/.ros/log/bbe0d8be-780d-11e5-930e-3ca9f41e22f4/zed_wrapper_node-2.log].
log file: /home/matt/.ros/log/bbe0d8be-780d-11e5-930e-3ca9f41e22f4/zed_wrapper_node-2*.log

Has anyone seen this before?  I looked for the camera in /dev/ but didn't find where it is coming in.

Thanks for your help,

Matt

closed 2015-11-06 by droter

rajkn90, 2015-11-20:
Even I am facing this issue. Did it work for you? What's the fix?

droter, 2015-11-21:
I haven't been able to get it working yet.