stereolabs/zed-ros-wrapper #10: ZED utilities work without GPU, why doesn't this code?

AdamGlass, 2015-12-06:
So a number of the ZED utilities including the Explorer and the recorder work without a GPU.  There is clearly support for getting an image via CPU in the header file.  I don't see why the GPU is involved here at all -- it's just a camera.  I don't want ZED processing the data at all.

Myzhar, 2015-12-06:
You can use ZED without GPU processing.
Simply acquire the data stream as a couple of synchronized images not
rectified.

I made this on my laptop that does not have a Nvidia video card and I
processed data using OpenCV.

The only difference is that you will lose the performances of the ZED SDK
and mainly you will lose the amazing depth map obtained thanks to the
Stereolabs' algorithms.

My simple app (not completed, it was only a test) uses OpenCV and Qt5.
You can get it here:
https://github.com/Myzhar/qt-jetson-zed-opencv-nosdk

Walt

AdamGlass, 2015-12-06:
Yes, that's the path I'm heading down.  After i filed the issue, i realized
that the basic Ubuntu apps saw the ZED as a basic camera rather than some
opaque magic device.  I then tried it out OpenCV like you did and it works
fine.  Going this path lets me keep the processing within a single
non-opaque stack.

thanks,
Adam

closed 2015-12-14 by adujardin

ashishk98, 2017-02-16:
What if I just take the images from the zed camera and process it in openCV without any Nvidia GPU. Will I be able to make a depth map out of it?

cuevhv, 2022-04-07:
> @ashishk98 Sure you can. ZED gives sync video streams like any other USB cameras. However, to compute the depth map in real-time requires Nvidia GPU and CUDA. If you catch left and right frames from opencv you can create depth image / point cloud.
> 
> https://docs.opencv.org/3.1.0/dd/d53/tutorial_py_depthmap.html https://github.com/opencv/opencv/blob/master/samples/python/stereo_match.py

Does Zed sync video streams are rectified? Is there any place where we can get the camera parameters, so we don't have to calibrate the camera?

Thanks!

Myzhar, 2022-04-09:
@cuevhv You posted an off-topic question in a very old closed issue. Please use the Github issue zone to report bugs or for feature requests. You can post questions to the [Stereolabs community forum](community.stereolabs.com) or you can write an email to support@stereolabs.com