stereolabs/zed-ros-wrapper #12: catkin_make fail on jetson tk1

sonyccd, 2016-01-02:
I can not seem to build the ROS wrapper on the Jetson TK1.  It is saying that it needs lubcudpp.so and I checked the zed libs and it is not there.  I did find it installed and ROS worked on my desktop machine.

`[ 50%] Built target zed_wrapper_gencfg
make[2]: *** No rule to make target `/usr/local/zed/lib/libcudpp.so', needed by `/home/ubuntu/Documents/ros_ws/ros_zed/devel/lib/zed_wrapper/zed_wrapper_node'.  Stop.
make[1]: *** [zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

adujardin, 2016-01-02:
Hi, 
You can solve this by editing the file '/usr/local/zed/zed-config.cmake'. 
Simply remove the 2 lines "/usr/local/zed/lib/libcudpp.so" and "/usr/local/zed/lib/libcudpp_hash.so".

The ZED SDK for jetson doesn't need libcudpp anymore but the desktop version still does, and it's the same cmake configuration file for both. This should be fixed in the next release.

Let me know if this fixes the problem, I can't test it at the moment.

sonyccd, 2016-01-02:
That worked thank you.

closed 2016-01-02 by sonyccd

rnunziata, 2016-01-24:
I have tried this and get the following error on my Jetson tk1:

Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node
/usr/local/zed/lib/libsl_zed.so: undefined reference to `XInitThreads'

Any ideas on how to proceed....am I missing a package? I did install the zed sdk. I am kind of stuck at this point.