stereolabs/zed-ros-wrapper #13: unable to compile zed wrapper on jetson tk1

rnunziata, 2016-01-24:
I am trying to install zed_wrapper_node for ros on my jetson tk1.
After installing libopencv4tegra and libopencv4tegra-dev

I was unable to install cv_bridge and image_transport because the installer wanted to uninstall the opencv4tegra lbs.

So I downloaded the source for them ros vision_opencv and image_common and compiled them.

I removed "/usr/local/zed/lib/libcudpp_hash.so" and "/usr/local/zed/lib/libcudpp.so"  from'/usr/local/zed/zed-config.cmake' as is recommend in a blog post that these are needed only for zed on the desktop.

Now I am left with following error:

Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node
/usr/local/zed/lib/libsl_zed.so: undefined reference to `XInitThreads'

What is the issue or issues ... Is zed ros wrapper not for the jetson tk1?

Myzhar, 2016-01-25:
Which version of linux4tegra are you using on Jetson TK1?
As far as I know with the latest version installed using "Jetpack 2" the problem with the conflict between OpenCV and OpenCV4Tegra has been solved.

About your linking problem try adding "-lX11" to your linking stage adding X11 to "target_link_libraries"...

rnunziata, 2016-01-25:
The zed sdk I've installed is ZED_SDK_Linux_JTK1_v0.9.2_alpha

I updated my system using the following from jetsonhacks site

http://developer.download.nvidia.com/embedded/OpenCV/L4T_21.1/libopencv4tegra-repo_l4t-r21_2.4.10.1_armhf.deb
sudo dpkg -i libopencv4tegra-repo_l4t-r21_2.4.10.1_armhf.deb
sudo apt-get update
sudo apt-get install libopencv4tegra libopencv4tegra-dev

cv_brigde when installing via the ubuntu software manager or via apt-get
install I get libopencv4terga will be removed
This is also true if I try to install indigo-cv-camera and
indigo-image-geometry on the jetson via the software manager.

 I included the X11 in zed-ros-wapper CMakeList and that worked and have
successfully run zed depth launch file.
Though for some reason I had to runt the catkin_make twice the first time
having failed with:make[1]: *** [vision_opencv/cv_bridge/src/
CMakeFiles/cv_bridge_boost.dir/all] Error 2

 It would be nice if the zed-ros-wapper github could be updated to have a
jetson branch.

If there is anything else I can do help with why the software manager is
finding conflicts just ask.

Thank you for your help
Richard Nunziata

Myzhar, 2016-01-25:
It is a known issue with OpenCV4Tegra and cv_bridge dependencies.
You can read more about it on devtalk, the official Nvidia forum.

A lot of people tried to solve it... and I got success only installing the
whole system using Jetpack v2

The only way to get everything working without using Jetpack v2 is
compiling OpenCV by yourself and then installing cv_bridge. In this way you
will remove the missing opencv dependencies and you have cv_bridge fully
working.
The only issue is that you will lose the optimization made by Nvidia for
Tegra core that are available using OpenCV4Tegra.

closed 2016-02-01 by rnunziata