stereolabs/zed-ros-wrapper #14: catkin_make problem

won13y, 2016-01-26:
I tried to catkin_make ZED-ros-wrapper,

and I already installed ZED_SDK_Linux_JTK1_v0.9.2_alpha.run sucessfully,

and dependency is also satistifed.

it is related to "No rule to make target `/usr/local/zed/lib/libcudpp.so"

ubuntu@tegra-ubuntu:~/catkin_ws$ cm

    make[2]: *** No rule to make target `/usr/local/zed/lib/libcudpp.so', needed by `/home/ubuntu    /catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node'.  Stop.
    make[2]: *** Waiting for unfinished jobs....
    [100%] Building CXX object zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir    /src/zed_wrapper_node.cpp.o
    Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/opencv_ros/opencv_ros_node
    make[2]: warning:  Clock skew detected.  Your build may be incomplete.
    [100%] Built target opencv_ros_node
    make[1]: *** [zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
    make: *** [all] Error 2

    Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/jetsonbot/devel;/home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/jetsonbot/devel;/home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - camera_umd (metapackage)
-- ~~  - opencv_tests
-- ~~  - image_geometry
-- ~~  - vision_opencv (metapackage)
-- ~~  - cv_bridge
-- ~~  - jpeg_streamer
-- ~~  - opencv_ros
-- ~~  - opencv_apps
-- ~~  - urg_node
-- ~~  - uvc_camera
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'camera_umd'
-- ==> add_subdirectory(camera_umd/camera_umd)
-- +++ processing catkin package: 'opencv_tests'
-- ==> add_subdirectory(vision_opencv/opencv_tests)
-- +++ processing catkin package: 'image_geometry'
-- ==> add_subdirectory(vision_opencv/image_geometry)
-- +++ processing catkin metapackage: 'vision_opencv'
-- ==> add_subdirectory(vision_opencv/vision_opencv)
-- +++ processing catkin package: 'cv_bridge'
-- ==> add_subdirectory(vision_opencv/cv_bridge)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   python
-- +++ processing catkin package: 'jpeg_streamer'
-- ==> add_subdirectory(camera_umd/jpeg_streamer)
-- Boost version: 1.54.0
[... 165 lines cut ...]
[ 75%] Building CXX object opencv_ros/CMakeFiles/opencv_ros_node.dir/src/image_converter.cpp.o
Built target urg_node
[ 75%] Built target camshift_exe
[ 76%] Built target convex_hull_exe
[ 76%] Built target contour_moments_exe
[ 76%] Built target edge_detection_exe
[ 76%] [ 78%] Built target fback_flow_exe
Built target face_detection_exe
[ 78%] Built target find_contours_exe
[ 79%] Built target general_contours_exe
[ 80%] Built target goodfeature_track_exe
[ 82%] Built target hough_circles_exe
[ 82%] Built target hough_lines_exe
[ 83%] Built target lk_flow_exe
[ 83%] Built target opencv_apps_generate_messages
[ 83%] Built target people_detect_exe
[ 84%] [ 95%] Built target opencv_apps
Built target phase_corr_exe
[ 95%] [ 95%] Built target segment_objects_exe
Built target simple_compressed_example_exe
[ 97%] Built target simple_example_exe
Scanning dependencies of target zed_wrapper_node
[ 97%] Built target simple_flow_exe
[ 98%] Built target watershed_segmentation_exe
ZED_SDK_Linux_JTK1_v0.9.2_alpha

adujardin, 2016-01-26:
Hi, yes this is a known bug on the tk1 with 0.9.2. A similar issue (#12) has been posted where you can find the solution.

closed 2016-01-26 by adujardin

sumitjagdale, 2017-03-28:
Hello,
I am facing the similar problem,  even I tried solution at #12 too but it did not work for me:(
Many thanks!
#########################################################
roszed@roszed-X555LN:~/zed_ws$ catkin_make 
Base path: /home/roszed/zed_ws
Source space: /home/roszed/zed_ws/src
Build space: /home/roszed/zed_ws/build
Devel space: /home/roszed/zed_ws/devel
Install space: /home/roszed/zed_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/roszed/zed_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/roszed/zed_ws/build"
####
Scanning dependencies of target zed_wrapper_gencfg
[ 33%] Generating dynamic reconfigure files from cfg/Zed.cfg: /home/roszed/zed_ws/devel/include/zed_wrapper/ZedConfig.h /home/roszed/zed_ws/devel/lib/python2.7/dist-packages/zed_wrapper/cfg/ZedConfig.py
../catkin_generated/env_cached.sh: 16: exec: /home/roszed/zed_ws/src/zed-ros-wrapper-master/cfg/Zed.cfg: Permission denied
make[2]: *** [/home/roszed/zed_ws/devel/include/zed_wrapper/ZedConfig.h] Error 126
make[1]: *** [zed-ros-wrapper-master/CMakeFiles/zed_wrapper_gencfg.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
roszed@roszed-X555LN:~/zed_ws$ sudo catkin_make
[sudo] password for roszed: 
sudo: catkin_make: command not found
roszed@roszed-X555LN:~/zed_ws$ catkin_make zed-ros-wrapper
Base path: /home/roszed/zed_ws
Source space: /home/roszed/zed_ws/src
Build space: /home/roszed/zed_ws/build
Devel space: /home/roszed/zed_ws/devel
Install space: /home/roszed/zed_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/roszed/zed_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/roszed/zed_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/roszed/zed_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/roszed/zed_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/roszed/zed_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper-master)
*** On Ubuntu 14, change line 30 of CMakeLists.txt to CUDA 7.5
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing:  _RSSDK_LIBRARIES) 
** WARNING ** visualization features related to rssdk will be disabled
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SURFACE
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_2D
-- looking for PCL_FEATURES
-- looking for PCL_GEOMETRY
-- looking for PCL_VISUALIZATION
-- looking for PCL_OUTOFCORE
-- looking for PCL_ML
-- looking for PCL_SEGMENTATION
-- looking for PCL_REGISTRATION
-- looking for PCL_RECOGNITION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_STEREO
-- looking for PCL_PEOPLE
-- Using these message generators: gencpp;genlisp;genpy
-- Configuring done
-- Generating done
-- Build files have been written to: /home/roszed/zed_ws/build
####
#### Running command: "make zed-ros-wrapper -j4 -l4" in "/home/roszed/zed_ws/build"
####
make: *** No rule to make target `zed-ros-wrapper'.  Stop.
Invoking "make zed-ros-wrapper -j4 -l4" failed