stereolabs/zed-ros-wrapper #16: zed_stereo.launch doesn't work on jetson TK1

mos127, 2016-01-28:
Hi,

zed_depth.launch works perfectly fine but zed_stereo.launch doesn't: this is the screen output:

```
... logging to /home/ubuntu/.ros/log/561992e4-c562-11e5-a71f-b8aeed732665/roslaunch-tegra-ubuntu-3352.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.2:45186/

SUMMARY
========

PARAMETERS
 * /camera/zed_wrapper_node/frame_rate: 15
 * /camera/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /camera/zed_wrapper_node/left_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/left_topic: left/image_raw
 * /camera/zed_wrapper_node/quality: 1
 * /camera/zed_wrapper_node/resolution: 2
 * /camera/zed_wrapper_node/second_cam_info_topic: right/camera_info
 * /camera/zed_wrapper_node/second_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/second_topic: right/image_raw
 * /camera/zed_wrapper_node/sensing_mode: 1
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)
  /
    zed_base_link (tf/static_transform_publisher)

ROS_MASTER_URI=http://192.168.0.1:11311

core service [/rosout] found
process[zed_base_link-1]: started with pid [3361]
process[camera/zed_wrapper_node-2]: started with pid [3362]
[ INFO] [1453960811.714012251]: ZED_WRAPPER Node initialized
[ INFO] [1453960811.741991418]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GK20A, ID : 0 
ZED SDK >> (Init) Quality mode has been set to None -- no disparity map
ZED SDK >> (Init) Creating ZED GPU mem...
[ INFO] [1453960811.927280585]: NO_GPU_COMPATIBLE
[camera/zed_wrapper_node-2] process has died [pid 3362, exit code 1, cmd /home/ubuntu/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node 0 __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/561992e4-c562-11e5-a71f-b8aeed732665/camera-zed_wrapper_node-2.log].
log file: /home/ubuntu/.ros/log/561992e4-c562-11e5-a71f-b8aeed732665/camera-zed_wrapper_node-2*.log
```
What can be the solution to this?!

Thanks

adujardin, 2016-01-28:
Hi, 
Unfortunately, this is a bug caused by the disparity mode NONE. 
The only solution right know is to set the disparity mode to PERFORMANCE, even if you don't use it.
This bug has been identified and will be fixed in the next release. Sorry about that.

mos127, 2016-01-28:
You mean by setting "quality" to "PERFORMANCE"?
Doesn't help with this solution even!

mos127, 2016-01-28:
What I did, I change this line (line 227) in the "zed_wrapper_node.cpp" : 
```
        err = zed->init(sl::zed::MODE::NONE, -1, true);
```
to 
```
        err = zed->init(sl::zed::MODE::PERFORMANCE, -1, true);
```
and now it works fine!

Thanks

adujardin, 2016-01-28:
Yes, that's what I meant.
Good, glad to hear it!

closed 2016-01-28 by adujardin

atomoclast, 2016-03-01:
I tried to change line 227 in the source code, couldn't find the line specified. The closest I could find was `err = zed->init(static_cast<sl::zed::MODE> (quality), -1, true);`

Tried to change it to no avail. Any suggestions?

adujardin, 2016-03-01:
The variable `quality` which sets the `sl::zed::MODE` is an argument set by the launch file, if you want you can change it [there](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed.launch#L11) or directly into the code (as in the comment above).
Does that address your issue?

atomoclast, 2016-03-01:
I tried to change the line:

`      <param name="quality"               value="2" />`

to 

`      <param name="quality"               value="PERFORMANCE" />`

I don't know if I understood you properly.

adujardin, 2016-03-01:
Ok, you have to put an integer in the launch file. So to get the `PERFORMANCE` mode you should put 1 `<param name="quality" value="1" />` (or 0 for NONE, [doc](https://www.stereolabs.com/developers/documentation/API/group__Enumerations.html#gaaa08b1713c3b6ca1930fbe3e5d35c831))

atomoclast, 2016-03-02:
I figured I'd have to have a different int. before trying "PERFORMANCE". I tried both "0" and "1" before hand with no luck.

this is what was in my Terminal:

`started roslaunch server http://ultron:58273/

SUMMARY
========

PARAMETERS
 * /camera/zed_wrapper_node/cloud_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /camera/zed_wrapper_node/depth_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/depth_topic: depth/image_rect_...
 * /camera/zed_wrapper_node/frame_rate: 30
 * /camera/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /camera/zed_wrapper_node/left_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/left_topic: left/image_rect_c...
 * /camera/zed_wrapper_node/point_cloud_topic: point_cloud/cloud
 * /camera/zed_wrapper_node/quality: 3
 * /camera/zed_wrapper_node/resolution: 2
 * /camera/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /camera/zed_wrapper_node/rgb_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /camera/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /camera/zed_wrapper_node/right_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/right_topic: right/image_rect_...
 * /camera/zed_wrapper_node/sensing_mode: 1
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)
  /
    zed_base_link1 (tf/static_transform_publisher)
    zed_base_link2 (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[zed_base_link1-1]: started with pid [11955]
process[zed_base_link2-2]: started with pid [11956]
process[camera/zed_wrapper_node-3]: started with pid [11957]
[ INFO] [1456943152.078205075]: ZED_WRAPPER Node initialized
[ INFO] [1456943152.097578424]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GeForce GTX 970, ID : 0 
ZED SDK >> (Init) Quality mode has been set to 
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Trying self calibration... 
[camera/zed_wrapper_node-3] process has died [pid 11957, exit code -11, cmd /home/hmi/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/hmi/.ros/log/1a90a430-e0a4-11e5-aed6-64006a646a5f/camera-zed_wrapper_node-3.log].
log file: /home/hmi/.ros/log/1a90a430-e0a4-11e5-aed6-64006a646a5f/camera-zed_wrapper_node-3*.log
`