stereolabs/zed-ros-wrapper #17: How to increase frame rate

shabhu18, 2016-02-11:
I am running zed on nvidia jetson TK1 . The frame rate which i am getting is around 4hz which is too low for processing real time algorithms . How do we increase the frame rate . 

please provide with optimal parameters .

[image]
[image]

adujardin, 2016-02-11:
Hi,

Well, depending on your application but several things can be done to speed it up:
- using VGA resolution
- using sl::zed::MODE::PERFORMANCE (not really an option on the TK1, they're both the same)
- using SENSING_MODE::RAW : it's faster and more precise but the occlusions are not filled. This mode should be used in most cases.
- the least CPU operations possible, prefer using the GPU function
- the least copy/retrieve CPU-GPU possible, the memory on the K1 is very slow. You should use [zero copy](http://arrayfire.com/zero-copy-on-tegra-k1/)
- disabling display, this prevent from computing unnecessary things like normalizeMeasure() or CPU retrieve to call imshow

Using ROS can reduce the performance, because of the additional processing required. For instance, the images have to be cast in RGB from RGBA in CPU.

Also, we're working on an update that will improve the performance.

shabhu18, 2016-02-11:
i guess most of the performance parameters are set up in wrapper itself . The frame rate are quite good when i run through sdk.once i switch on to ros everything is very slow.Maybe zed datastructure to opencv conversion problem (just guessing).

All the parameters i tuned to maximum performance . zero copy i am not aware too much how to it fit into the scenario. 
I cannot disable disparity ,that what i need as input . 
Gui all disabled .
 No external cpu / gpu program running.

adujardin, 2016-02-12:
I understand, those were general advice.

In case you want the best performance, I recommend not using ROS. ROS is a great tool but too generalised. As a result, the pipeline is not efficient, especially for the TK1.

However, we will look into this to find potential optimizations in the wrapper for the Jetson.

closed 2016-03-04 by adujardin

jeremy-a, 2016-03-07:
Since this issues is closed, does that mean the latest commit includes updates which improve framerate. What pointcloud frame rate should we expect using ROS on a TK1? I'm getting about 2Hz for my point cloud. I'm running the launch file from an ssh terminal from the TK1 and RVIZ on a separate laptop.

Are there any other runtime, startup, or envioronment setup items that could improve performance?

shabhu18, 2016-03-08:
I am facing the same issue .I migrated from BUMBLEBEE2 to ZED seeing their disparity was damm good. We cannot use big bulky gpu on robot . Their is some problem with data structure coping between opencv and zed .  
Waiting for a commit asap .

shabhu18, 2016-03-18:
What frame rate we get with TX1. Any ideas??

avirtuos, 2016-07-23:
I'm seeing 2HZ (point cloud) on the TX1 with the below config:

`<param name="resolution"            value="1" />
      <param name="quality"               value="1" />
      <param name="sensing_mode"          value="1" />
      <param name="frame_rate"            value="60" />
      <param name="odometry_DB"           value="" />
      <param name="openni_depth_mode"     value="0" />

      <param name="rgb_topic"             value="rgb/image_rect_color" />
      <param name="rgb_cam_info_topic"    value="rgb/camera_info" />
      <param name="rgb_frame_id"          value="/zed_tracked_frame" />

      <param name="left_topic"            value="left/image_rect_color" />
      <param name="left_cam_info_topic"   value="left/camera_info" />
      <param name="left_frame_id"         value="/zed_tracked_frame" />

      <param name="right_topic"           value="right/image_rect_color" />
      <param name="right_cam_info_topic"  value="right/camera_info" />
      <param name="right_frame_id"        value="/zed_tracked_frame" />

      <param name="depth_topic"           value="depth/image_rect_color" />
      <param name="depth_cam_info_topic"  value="depth/camera_info" />
      <param name="depth_frame_id"        value="/zed_tracked_frame" />

      <param name="point_cloud_topic"     value="point_cloud/cloud" />
      <param name="cloud_frame_id"        value="/base_frame" />

      <param name="odometry_topic"                value="odom" />
      <param name="odometry_frame_id"             value="/zed_initial_frame" />
      <param name="odometry_transform_frame_id"   value="/zed_tracked_frame" />`

avirtuos, 2016-07-25:
I just wanted to close the loop here incase others find this and are looking for an alternative. Since I mostly needed to get a 2D 'laser scan' from my ZED's depth image I found a way to avoid using the pointcloud, which after profiling the wrapper's C/++ code is def the bottleneck. by using http://wiki.ros.org/depthimage_to_laserscan I am able to get 18+ HZ for the laser scan created from the ZED's HD depth image. Though when enabling MEDIUM or HIGH quality the usefulness of the depth image for laser scans of 'rooms' drops considerably though object detection improves significantly.