stereolabs/zed-ros-wrapper #21: Point cloud crashes after first publish of pointcloud

eric-wieser, 2016-02-29:
After subscribing to the `point_cloud/image_rect` topic, I get exactly one message before the zed launch file dies

stdout:

```
SUMMARY
========

PARAMETERS
 * /camera/zed/zed_wrapper_node/frame_rate: 15
 * /camera/zed/zed_wrapper_node/left_cam_info_topic: rgb/camera_info
 * /camera/zed/zed_wrapper_node/left_frame_id: /zed_optical_frame
 * /camera/zed/zed_wrapper_node/left_topic: rgb/image_rect_color
 * /camera/zed/zed_wrapper_node/quality: 2
 * /camera/zed/zed_wrapper_node/resolution: 3
 * /camera/zed/zed_wrapper_node/second_cam_info_topic: depth/camera_info
 * /camera/zed/zed_wrapper_node/second_frame_id: /zed_optical_frame
 * /camera/zed/zed_wrapper_node/second_topic: depth_registered/...
 * /camera/zed/zed_wrapper_node/sensing_mode: 1
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/zed/
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)
  /
    zed_base_link1 (tf/static_transform_publisher)
    zed_base_link2 (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[zed_base_link1-1]: started with pid [16249]
process[zed_base_link2-2]: started with pid [16250]
process[camera/zed/zed_wrapper_node-3]: started with pid [16251]
[ INFO] [1456717432.735849340]: ZED_WRAPPER Node initialized
[ INFO] [1456717432.788498337]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GM20B, ID : 0 
ZED SDK >> (Init) Quality mode has been set to Quality mode
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Trying self calibration... 
ZED SDK >> (Init) Done...
[ INFO] [1456717437.278710947]: SUCCESS
[ INFO] [1456717439.328094330]: Reconfigure confidence : 80
[ INFO] [1456717439.566147235]: Advertized on topic rgb/image_rect
[ INFO] [1456717439.568182066]: Advertized on topic rgb/image_rect_color
[ INFO] [1456717439.570107002]: Advertized on topic right/image_rect
[ INFO] [1456717439.572015897]: Advertized on topic depth/image_rect
[ INFO] [1456717439.574672599]: Advertized on topic point_cloud/image_rect
[ INFO] [1456717439.576888419]: Advertized on topic rgb/camera_info
[ INFO] [1456717439.577281645]: Advertized on topic rgb/camera_info
[ INFO] [1456717439.578923198]: Advertized on topic right/camera_info
[ INFO] [1456717439.580648239]: Advertized on topic depth/camera_info
[camera/zed/zed_wrapper_node-3] process has died [pid 16251, exit code -11, cmd /home/racecar/racecar_team5_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/racecar/.ros/log/aa2acb02-de8d-11e5-b9a7-00044b580bc9/camera-zed-zed_wrapper_node-3.log].
log file: /home/racecar/.ros/log/aa2acb02-de8d-11e5-b9a7-00044b580bc9/camera-zed-zed_wrapper_node-3*.log
```

There seem to be no further log files

nesnes, 2016-02-29:
Does it happens all the time or is it a one time crash? Ccan you describe a bit more the circumstances so we may be able to reproduce the bug?

nesnes, 2016-02-29:
Ok! so it looks like you're not using the last zed.launch file provided in the update #19 . I can see it because your point_cloud topic doesn't have the expected name.
So please delete your zed_wrapper folder in your catkin workspace and download the last version of the repository. 
Run the `catkin_make clean` command from your workspace root.
And follow the small build steps in the README file.

From your output: 
`[ INFO] [1456717439.574672599]: Advertized on topic point_cloud/image_rect`

You should have
`[ INFO] [1456717439.574672599]: Advertized on topic point_cloud/cloud`

_If even after that you continue to have the issue, please come with all the informations you can get to help us to reproduce the crash_

eric-wieser, 2016-02-29:
Thanks, I'll give that a go. I was having difficulty getting the new launch file to work, but I think I was jumping to the wrong conclusions and zed had just gotten into a bad state

eric-wieser, 2016-02-29:
Still no dice

eric-wieser, 2016-02-29:
Apparently `-11` is `SIGSEGV` -  a segmentation fault. I think this is almost certainly a use-after-free introduced in #19.

adujardin, 2016-03-01:
Hi,
There is actually a bug in the current ZED SDK. 
If the state is not `SUCCESS`, the destructor will leak some (quite a lot actually) GPU memory when called.
This leads to a `SIGSEV` (out of memory in this case) after several calls, especially on the K1.

Sorry for the inconvenience. This should be fixed for the next release

eric-wieser, 2016-03-01:
The destructor of what? `Mat`? I don't recall that having a destructor, but maybe that's the point

adujardin, 2016-03-01:
The destructor of `Camera`

eric-wieser, 2016-03-01:
Hopefully that only gets called once though

adujardin, 2016-03-01:
Most of the cases yes. 
But if the ZED was disconnected for example, the destructor can be called more than once. (https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L412)

However, if the ZED was not unplugged your issue is probably related to something else

eric-wieser, 2016-03-01:
Yep, that's not the cause of my problem - I've seen the`"Wait for a new image to proceed"` warning on the previous line appear in my output in other tests, but it doesn't do so when this crash occurs.

eric-wieser, 2016-03-02:
To confirm - moving point cloud parsing into the main thread fixes the problem

adujardin, 2016-04-21:
Hi, 
Is this still an issue with the current version ?

closed 2016-06-24 by adujardin

eric-wieser, 2016-06-26:
I can no longer say - I've now lost access to the hardware