stereolabs/zed-ros-wrapper #23: unable to compile zed wrapper on jetson tk1 XinittTreads

rnunziata, 2016-03-02:
I  just downloaded the new zed ros wrapper ( though I did not a system general update via software center) and now I can not compile the wrapper even though I include X11 in the CMakeLists as I have done and worked before. Any ideas on how to proceed?

say libsel_zed.so  undefined reference to XInittTreads.

adujardin, 2016-03-07:
Hi, 

Could you check if the libsl_zed.so is able to find libX11.so : 
`ldd /usr/local/zed/lib/libsl_zed.so | grep libX11`
It should be found in `/usr/lib/arm-linux-gnueabihf/`

rnunziata, 2016-03-07:
no...it finds nothing. However it is the directory where it should be found.

sl-braun, 2016-03-07:
Hi,

If you don't have libX11.so installed in /usr/lib/arm-linux-gnueabihf/, then do 
sudo apt-get install libX11-dev

If you have libX11.so/so.6 already in that directory, check that you have the rights/permission to access it, or your ld configuration (etc/ld.so.conf.d/)

rnunziata, 2016-03-07:
Here are results please supply command

rnunziata, 2016-03-09:
I did the following as per your request  which has not fixed the issue.   I
do not wish to do random mods to my system. Has zed tested the camera with
the latest version / update of ubuntu 14.04 on the Jetson? If I can not get
this to work I spent a lot of money for something that will not meet my
needs. zed says it supports ROS integration with Jetson please advise I
have done nothing then follow to the letter the installation documents.

sudo chown :ubuntu libX11.so libX11.so.6

which resulted in the following

lrwxrwxrwx 1 root root         15 Dec  4  2013 libX11.so -> libX11.so.6.3.0
lrwxrwxrwx 1 root root         15 Apr  1  2015 libX11.so.6 ->
libX11.so.6.3.0
-rw-r--r-- 1 root ubuntu   905864 Apr  1  2015 libX11.so.6.3.0

On Mon, Mar 7, 2016 at 12:51 PM, Richard Nunziata <richardnunziata@gmail.com

rnunziata, 2016-03-10:
How can I cross compile the zed ros wrapper on my laptop....I want to see if this is a jetson issue, and if it works at least it will let me move forward.

rnunziata, 2016-03-13:
Question: is it possible that the xserver-org has changed even though I set it on hold as per blogs for Jetston TK1...should I take it off hold and do an update or reinstall. Maybe this could correct the issue.

rnunziata, 2016-03-13:
I was able to compile 3DViewer in samples/olg/gpu and run it. It did not ask for X11. So this appears to be a wrapper implementation / compilation issue.

rnunziata, 2016-03-16:
I took the hold off of xserver-xorg-core and updated/upgrade but I still get the same compile error.

rnunziata, 2016-10-03:
new release on jetpack seemed to fix this issue for me

closed 2016-10-03 by rnunziata