stereolabs/zed-ros-wrapper #25: depth image is very imprecise

eric-wieser, 2016-03-04:
In my testing, I only ever get integers in [0, 10] for the pixels of the depth image. Is this measured in meters?

Is the issue that this is being incorrectly rounded in [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L466)?

P-yver, 2016-03-04:
Hi Eric,
We convert the baseline from  millimeter to meter, see [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L230). So yes the depth values we publish are in meters.

P-yver, 2016-03-04:
It is not an issue, it is due to the conversion from float to short.
We use it in meter for display purpose. If you want you can apply a scale factor during the conversion function.

eric-wieser, 2016-03-04:
This is an issue, because when converting to short you throw away all the fractional meter information. There's nothing i can do in my code to recover this infomeation. Why not just leave the data as a float, as suggested by REP 118?

eric-wieser, 2016-03-04:
Also, I don't see how that line shows you publish meters. All you're doing there is changing the baseline that's reported to ROS. I don't see how that relates to the calculations done by the zed sdk

closed 2016-03-04 by adujardin

eric-wieser, 2016-03-04:
Fixed in #26

eric-wieser, 2016-03-07:
@P-yver: your comments are wrong, and this patch is incorrect. The zed camera returns distances in the same units as the distance between cameras is measured. By default, this is 120mm. Therefore, the depth measurements are returned in mm.

In a previous version of the readme, there were instructions to manually set this baseline to `0.12`. This instruction should no longer apply. Then everything works correctly, and is published to the nearest mm (which is fine).

syamprasadkr, 2016-09-16:
I am trying to obtain depth value of each pixel from the zed camera. I am trying to convert the ros image published in the topic 

> /camera/depth/image_rect_color 

to OpenCV image using [this](http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages).

I am interested in knowing what other approach is possible by just using the `zed-ros-wraper`. Can you pls elaborate your method if not inconvenient for you? Thank You.