stereolabs/zed-ros-wrapper #27: Process keeps dying on launch (Non-TK1 Nvidia GPU)

atomoclast, 2016-03-07:
Hello, I'm currently trying to use the Zed-ROS-Wrapper on my desktop with a GeForce GTX 970. As you can see in the attached file, it seems the find the GPU and almost launch, then crash while self-calibrating. 

 I have searched through github and seen that changing the 

<param name="quality"               value="2" /> to "1" in the zed.launch file has helped (at least for the TK1). We still are experiencing issues.  

Please see attached screenshot. 

[image]

adujardin, 2016-03-08:
Hi,

I am not able to reproduce the problem. Does the issue occur without the ROS wrapper, with the other samples?

The only solution, for now, is to disable self-calibration. You can do that by changing [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L323) to : 
`err = zed->init(static_cast<sl::zed::MODE> (quality), -1, true /*verbose*/, false /*vflip*/, true /* disable auto calib */);`

atomoclast, 2016-03-08:
Hello,

I have change the line of code to disable self-calibration, I am still getting a process has died error. Please see the attached screenshot.

[image]

adujardin, 2016-03-08:
Do you have the issue with the other samples / tools?

It seems that the ZED was not found during the first part of the init process. Do you have random disconnection when using the ZED? Maybe try other USB ports.

Also, could you try with an SVO file?

atomoclast, 2016-03-08:
@adujardin 

I was able to to run the ZED Depth Viewer last week and a few other tools like the Explorer with no issues every other time before that. I just tried to launch it right now, and it seems to be stuck in a self-calibration loop now...maybe I need to reinstall the ZED packages?

I tried to pull the latest version of the wrapper for ROS yesterday to see if that was the issue. Maybe I need to resetup the ZED package.

atomoclast, 2016-03-08:
I have tried to reinstall the ZED SDK packages, it still is stuck in a calibration loop. 

I am however able to open the Zed via a quick openCV script that opens it as  capture device, so the computer is reading it...

adujardin, 2016-03-16:
Hi,
Sorry for the delay. Could you try the following:

1. Install the latest version of the ZED SDK [v0.9.2c](https://www.stereolabs.com/download_327af3/ZED_SDK_Linux_x86_64_v0.9.2c_beta.run)

2. Plug your ZED and launch `/usr/local/zed/tools/ZED\ Settings\ App -factory` to download your calibration file (even if you already have the conf file, the calibration process slightly change recently, the tools must be up to date). If you have a proxy/firewall issue, you can download your file at `calib.stereolabs.com/?SN={ZED_SERIAL_NUMBER}` and place it in `/usr/local/zed/settings/`

3. Launch a sample or `ZED Depth Viewer`, if it fails:

4. USB3 check: launch the `ZED Explorer` and test every resolution. If you see corrupted green/purple frames, it means that your USB3 is not fast enough. Try another port (avoids front port, hub, extension cord...)

Let me know the results

atomoclast, 2016-03-16:
Hello, no problem. It tried to reinstall everything and the ZED Depth Viewer seems to be working with every resolution now. So that's a step in the right direction. 

I did try to catkin_make the ROS wrapper and tried to launch it with both the quality set to 1 or 2. Still the same issue.

Thanks in advance!

[image]

adujardin, 2016-03-16:
OK, nice to see that `ZED Depth Viewer` is working correctly. 

Could you reactivate the auto calib in the ROS node?
`err = zed->init(static_cast<sl::zed::MODE> (quality), -1, true, false, false /* disable auto calib */);`

Also, could you try another sample, OpenCV for instance. It's to check if it's a ROS-related error. The error seems to happen during the initialization of the rectification, but it's the same process in `ZED Depth Viewer` and the other samples.

atomoclast, 2016-03-16:
I tried to change the line to what you had above and rebuilt the package. Still the same issue. I don't think it's ROS-related. We currently run a number of other things on ROS with no issues. That I know of.

For the sake of insanity, I even tried to go back to the original line of code when you download it. Still no luck.

adujardin, 2016-03-16:
What about the OpenCV sample, does it work?

What I meant by ROS-related error is that catkin may link with another version of a lib or do something that creates a conflict with the dependencies of the libsl_zed. Could you try the `ldd` command on the ROS node executable and on the `libsl_zed.so`?

atomoclast, 2016-03-16:
Which OpenCV sample? I looked through the /usr/local/zed directory and the examples and didn't see one specifically called "OpenCV". Unless you mean a sample that uses OpenCV....

Also I ran ldd commands and got this:

`hmi@ultron:/usr/local/zed/lib$ ldd libsl_zed.so 
	linux-vdso.so.1 =>  (0x00007ffff5bf3000)
	libcudart.so.6.5 => /usr/local/cuda-6.5/lib64/libcudart.so.6.5 (0x00007f651e560000)
	libsl_calibration.so => /usr/local/zed/lib/libsl_calibration.so (0x00007f651e26e000)
	libsl_depthcore.so => /usr/local/zed/lib/libsl_depthcore.so (0x00007f651ddc0000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007f651d989000)
	libopencv_highgui.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4 (0x00007f651d73e000)
	libopencv_imgproc.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4 (0x00007f651d2ae000)
	libopencv_calib3d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4 (0x00007f651d018000)
	libnppi.so.6.5 => /usr/local/cuda-6.5/lib64/libnppi.so.6.5 (0x00007f6518b75000)
	libnpps.so.6.5 => /usr/local/cuda-6.5/lib64/libnpps.so.6.5 (0x00007f651832a000)
	libusb-0.1.so.4 => /lib/x86_64-linux-gnu/libusb-0.1.so.4 (0x00007f6518121000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6517e1d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6517b17000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6517901000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f651753c000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6517338000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f651711a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f6516f12000)
	libopencv_video.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4 (0x00007f6516cbe000)
	libcudpp.so => /usr/local/zed/lib/libcudpp.so (0x00007f650caad000)
	libcuda.so.1 => /usr/lib/x86_64-linux-gnu/libcuda.so.1 (0x00007f650b88f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f650b676000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f650b33c000)
	libtbb.so.2 => /usr/lib/libtbb.so.2 (0x00007f650b108000)
	libjpeg.so.8 => /usr/lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f650aeb3000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f650ac8d000)
	libtiff.so.5 => /usr/lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f650aa1b000)
	libjasper.so.1 => /usr/lib/x86_64-linux-gnu/libjasper.so.1 (0x00007f650a7c4000)
	libIlmImf.so.6 => /usr/lib/x86_64-linux-gnu/libIlmImf.so.6 (0x00007f650a515000)
	libHalf.so.6 => /usr/lib/x86_64-linux-gnu/libHalf.so.6 (0x00007f650a2d2000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f6509c95000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f65099e2000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f6509791000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f6509489000)
	libgtkglext-x11-1.0.so.0 => /usr/lib/libgtkglext-x11-1.0.so.0 (0x00007f6509285000)
	libgdkglext-x11-1.0.so.0 => /usr/lib/libgdkglext-x11-1.0.so.0 (0x00007f6509021000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007f6508dad000)
	libv4l1.so.0 => /usr/lib/x86_64-linux-gnu/libv4l1.so.0 (0x00007f6508ba7000)
	libavcodec.so.54 => /usr/lib/x86_64-linux-gnu/libavcodec.so.54 (0x00007f6507e53000)
	libavformat.so.54 => /usr/lib/x86_64-linux-gnu/libavformat.so.54 (0x00007f6507b31000)
	libavutil.so.52 => /usr/lib/x86_64-linux-gnu/libavutil.so.52 (0x00007f650790c000)
	libswscale.so.2 => /usr/lib/x86_64-linux-gnu/libswscale.so.2 (0x00007f65076c5000)
	libopencv_flann.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4 (0x00007f650745b000)
	libopencv_features2d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4 (0x00007f65071b8000)
	libnppc.so.6.5 => /usr/local/cuda-6.5/lib64/libnppc.so.6.5 (0x00007f6506f5c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f651ea04000)
	libnvidia-tls.so.361.28 => /usr/lib/x86_64-linux-gnu/tls/libnvidia-tls.so.361.28 (0x00007f6506d58000)
	libnvidia-glcore.so.361.28 => /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.361.28 (0x00007f65050ca000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f6504d95000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f6504b83000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f6504961000)
	libjbig.so.0 => /usr/lib/x86_64-linux-gnu/libjbig.so.0 (0x00007f6504753000)
	libIex.so.6 => /usr/lib/x86_64-linux-gnu/libIex.so.6 (0x00007f6504535000)
	libIlmThread.so.6 => /usr/lib/x86_64-linux-gnu/libIlmThread.so.6 (0x00007f650432f000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f650412b000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f6503f1e000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f6503d18000)
	libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f6503af6000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f65037eb000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f65035ca000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f6503257000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f6503042000)
	libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f6502df5000)
[... cut ...]
	libnih.so.1 => /lib/x86_64-linux-gnu/libnih.so.1 (0x00007f64fa660000)
	libnih-dbus.so.1 => /lib/x86_64-linux-gnu/libnih-dbus.so.1 (0x00007f64fa456000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f64fa211000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f64fa00c000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f64f9e07000)
hmi@ultron:/usr/local/zed/lib$ ldd /home/hmi/catkin_ws/devel/lib/zed_wrapper/
ldd: /home/hmi/catkin_ws/devel/lib/zed_wrapper/: not regular file
`

adujardin, 2016-03-16:
For the sample, you can try [this one](https://github.com/stereolabs/zed-opencv) (the same one is located in `samples/cpu/with OpenCV`)

How did you install OpenCV? From the `ldd` result, I assume that you didn't recompile it?

Could you relaunch ldd on the following:
 - `ldd ~/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node`
 - and also:` ldd /usr/local/zed/tools/ZED\ Depth\ Viewer`

atomoclast, 2016-03-17:
I tried to build and launch the ZED with OpenCV example. (I didn't see that earlier, I was looking in the GUI folder assuming both folders had the same samples, just one for integrated graphics and the other for dedicated graphics.)

This was the output:

[image]

The outputs of the ldd commands are in the files below:

[ZEDdepthviewerLDDoutput.txt](https://github.com/stereolabs/zed-ros-wrapper/files/178503/ZEDdepthviewerLDDoutput.txt)
[ZEDwrapperLDDoutput.txt](https://github.com/stereolabs/zed-ros-wrapper/files/178504/ZEDwrapperLDDoutput.txt)

adujardin, 2016-03-17:
OK Thanks!

That's definitely not normal, it's the same code behind the samples and ZED Depth Viewer. It must be related to a dependency / configuration issue. 

The samples in the SDK has a precompiled version in the bin folder (`/usr/local/zed/sample/cpu/with\ OpenCV/bin/`), could you try this one? If this works could you ldd the precompiled version and the version you recompiled pls?

Could you test recompiling and installing [OpenCV 2.4.12](https://github.com/Itseez/opencv/archive/2.4.12.zip) and then recompiling this sample. Make sure that this version is taken by the sample, `ldd ZED\ with\ OpenCV | grep libopencv_core.so.2.4` should point to `/usr/local/lib/`

I suspect that the version you have is somewhat not compatible with the one we used (compilation option maybe).

Sorry, I know that's a lot of steps to try, but it's the first time we encountered this issue and we are not able to reproduce it.

atomoclast, 2016-03-18:
So the prebuilt version ran flawlessly. Here's the output for the ldd command: 

[PrebuiltOpenCVldd.txt](https://github.com/stereolabs/zed-ros-wrapper/files/179953/PrebuiltOpenCVldd.txt)

I rebuilt the example, it launched properly. SO then I tried to rerun catkin_make on the wrapper code.

It also ran flawlessly. I wonder if it was because we were running an older version of openCV2.4.

To be honest, since this is a lab computer, lots of changes happen sometimes. 

At least it works now! 

Thanks!

adujardin, 2016-03-18:
Ok Glad to hear your problem is solved!

closed 2016-03-18 by adujardin