stereolabs/zed-ros-wrapper #28: ZED NOT AVAILABLE On Ubuntu

xbklairith, 2016-03-14:
Environment On GigabyteBrix *(GB-BXi7G3-760)
CPU i7-4710HQ
GPU NVIDIA GeForce GTX 760 

> [ INFO] [1457929521.312857351]: Using ZED Camera
> ZED SDK >> (Init) Best GPU Found : GeForce GTX 760, ID : 0 
> ZED SDK >> (Init) Quality mode has been set to Performance mode
> [ INFO] [1457929521.860688457]: ZED_NOT_AVAILABLE
> ZED SDK >> (Init) Best GPU Found : GeForce GTX 760, ID : 0 
> ZED SDK >> (Init) Quality mode has been set to Performance mode
> [ INFO] [1457929524.325564349]: ZED_NOT_AVAILABLE

i have try following command ,
`luvcview -d /dev/video0 -s 2560x720 -i 15` 
*camera is working.*

Is there problems about SDK?

adujardin, 2016-03-14:
Hi,

Could you test if the other samples and tools work? You can try ZED Explorer, ZED Depth Viewer and OpenCV sample for example.

xbklairith, 2016-03-14:
Hi,

Depth viewer said "No ZED detected : Only ofline mode is available (with SVO files)"

adujardin, 2016-03-14:
OK, in that case, could you try the following commands in a terminal:
- `lsusb | grep 2b03:f580 | wc -l`

and also

- `ls /usr/local/zed/settings/`

xbklairith, 2016-03-15:
lsusb...  output is `1` 

/usr/local/zed/settings/ is empty

amiltonwong, 2016-03-15:
I have the same issue: https://github.com/stereolabs/zed-ros-wrapper/issues/29#issuecomment-196704282

adujardin, 2016-03-15:
`/usr/local/zed/settings/` should contains the calibration file for the camera (something like `SN{YourZEDSerialNumber}.conf`). The ZED SDK can't launch without it.

This file is normally downloaded at installation. However, you can do it with the tool `ZED Settings App` (Make sure you downloaded and installed the latest version of the SDK).
The ZED must be plugged in and you need an internet connection (no proxy or firewall).
Then launch `/usr/local/zed/tools/ZED\ Settings\ App -factory`

Or you can provide the serial number of the ZED (printed on the cable or with ZED Explorer) with 
`/usr/local/zed/tools/ZED\ Settings\ App -sn {YourZEDSerialNumber}`

If it works, `/usr/local/zed/settings/` should contains the calibration file.
If that doesn't work, let me know (I will need your serial number)

amiltonwong, 2016-03-15:
[image]

Hi, @adujardin ,
I follow the command : /usr/local/zed/tools/ZED\ Settings\ App -sn 000002532
And the calibration file is downloaded in /usr/local/zed/settings.
However, when I run command: "./ZED\ Depth\ Viewer"

milton@milton-OptiPlex-9010:/usr/local/zed/tools$ ./ZED\ Depth\ Viewer

I got the screen above, which shows me the ZED camera is not detected. Any other steps to troubleshoot it?
BTW, I already plug it into usb 3.0 port in my system. Is it required to enable usb 3.0 function in ubuntu 14.04? (If yes, how should i do?) Or any other command to verify the plugged port is in usb 3.0 function?

THX~

adujardin, 2016-03-15:
@amiltonwong 

Just to make sure, please verify that only one program at a time uses the ZED, for example if `ZED Settings App` is open, `ZED Depth Viewer` won't work. 

Could you try the ZED Explorer? Also does luvcview work `luvcview -d /dev/video0 -s 2560x720 -i 30`?

Even if the resolution and framerate are limited, the ZED can work on USB2.0 port. 
That's actually something you can test, plug your ZED in a USB2.0 port and launch it in VGA (`luvcview -d /dev/video0 -s 1280x480 -i 15`), if that work test a ZED tool. 

The problem might be related to a USB3 driver issue.
`lsusb | grep "Linux Foundation 3.0 root hub"` will tell you if Linux recognized a USB3 controller.

(However, I've seen controller that was supposed to be compatible with Ubuntu, recognized by `lsusb`, usable on Windows but ended up not working on Linux)

xbklairith, 2016-03-15:
After launch /usr/local/zed/tools/ZED\ Settings\ App -factory
Alert box :  

```
The ZED Camera has not been detected during setup. The required calibration file has not been downloaded.
After restart, please launch the ZED Settings App with -factory parameter
```

```

After use 
/usr/local/zed/tools/ZED\ Settings\ App -sn {YourZEDSerialNumber}
```
there is .conf file in `/usr/local/zed/settings/`

However, It is still not working

adujardin, 2016-03-15:
During the ZED SDK installation, did you choose 'auto install dependencies' ?

It seems that the ZED is working but the serial number can't be read. Could you open ZED Explorer and take a screenshot of it, please?

xbklairith, 2016-03-16:
Yes, There is serial number show in app.
[image]

amiltonwong, 2016-03-17:
Hi, @adujardin,

I found some clues. After I plug the Stereo camera into usb port. In my ubuntu system, I could find /dev/video0.
When I try the ZED Explorer, I got the following error telling me the ZED Explorer trying to use  '/dev/video-1'. I think that's the problem which troubles me. Any suggestion to fix it?

=====================================

milton@milton-OptiPlex-9010:/usr/local/zed/tools$ ./ZED\ Explorer 
Cannot identify '/dev/video-1': 2, No such file or directory
*** buffer overflow detected ***: ./ZED Explorer terminated
failed Bad file descriptor. (System code: 9) 
failed Bad file descriptor. (System code: 9) 
VIDIOC_G_PARM - Unable to get timeperframe: Bad file descriptor

=====================================================

amiltonwong, 2016-03-17:
Using temporary solution: create a soft link
>> cd /dev
>> sudo ln -s video0 video-1
And then run  ZED Explorer again, now I could see the blue led indicator shows up. But the ZED explorer only shows a dark screen, and that user interface responds very slowly:
[image]

The ID shows '-1', what does it mean?

nuno-or, 2016-04-08:
zed tools work on ubuntu 14.04 with older linux kernel 3.13

zed tools do **not** work on ubuntu 14.04.x (e.g. 14.04.4) with newer linux kernel 4.2.x

I also tried with ubuntu 14.04 and with the current latest linux kernel (4.5.0) and the zed tools do not work either.

I can see the zed camera using lsusb and can also use something like lucview as in the comment above to see the zed camera video stream on all the above ubuntu/linux kernel versions, so the system is able to access it, but not with either the zed tools or zed sdk.

nuno-or, 2016-04-08:
gdb backtrace of 'ZED Depth Viewer' running on ubuntu 14.04.4 , linux kernel 4.5.0 , zed sdk 0.9.3-beta:

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff790e636 in sl::zed::LinuxStereoGrabber::getZEDSerialNumber() () from /usr/local/zed/lib/libsl_zed.so
(gdb) bt
#0  0x00007ffff790e636 in sl::zed::LinuxStereoGrabber::getZEDSerialNumber() () from /usr/local/zed/lib/libsl_zed.so
#1  0x00007ffff78fe986 in sl::zed::Camera::openCamera(bool) () from /usr/local/zed/lib/libsl_zed.so
#2  0x00007ffff7902896 in sl::zed::Camera::init(sl::zed::MODE, int, bool, bool, bool) () from /usr/local/zed/lib/libsl_zed.so
#3  0x0000000000446b89 in InputThreadModul::live() ()
#4  0x0000000000444b01 in MainProcess::CreateLiveInput() ()
#5  0x0000000000452e63 in MainProcess::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#6  0x0000000001098fe7 in QMetaObject::activate(QObject*, int, int, void**) ()
#7  0x0000000000453931 in GUI::changeInput(int) ()
#8  0x000000000044aa08 in GUI::changeRes(int) ()
#9  0x0000000000453b3b in GUI::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#10 0x0000000001098fe7 in QMetaObject::activate(QObject*, int, int, void**) ()
#11 0x00000000005d81cc in QComboBoxPrivate::_q_emitCurrentIndexChanged(QModelIndex const&) ()
#12 0x00000000005d9bb4 in QComboBoxPrivate::setCurrentIndex(QModelIndex const&) ()
#13 0x00000000005d9e1f in QComboBox::setCurrentIndex(int) ()
#14 0x00000000004444d1 in MainProcess::MainProcess(int, char**) ()
#15 0x00000000004339b7 in main ()

nuno-or, 2016-04-13:
I did a git bisect on the linux kernel source tree and found the commit that makes the zed sdk fail. This particular commit changes some things in v4l2. I'm still investigating it, but at least now I have some clues where to dig in.

atomoclast, 2016-04-13:
Hello, I am currently experiencing the same issue myself. We recently got a new PC that we are using for some of our vision applications, and we are currently trying to port over all of our software and setup. 

During the setup of the ZED, I keep experiencing issues with it being able to connect to the internet. it says that it can connect to the ZED, but no internet connection (I can browse through a browser and ping through terminal though). I did try to suggested code above of: 

`/usr/local/zed/tools/ZED\ Settings\ App -sn {YourZEDSerialNumber}`

because I kept getting the same error when I tried to run:

`/usr/local/zed/tools/ZED\ Settings\ App -factory` and got the same error as a user above.

I do have a .conf file in the settings folder. It just won't open up the Depth Viewer or any of the other tools.

I have no idea what's going on or why it was so painless when I installed it last.

Thanks in advance!

nuno-or, 2016-04-13:
Hi, it seems the zed sdk gets confused when calling v4l_enum_fmt() and the description fields gets populated.  I found a workaround by patching the v4l2 linux kernel driver to avoid doing that. I'll post it soon for 4.2.0 (latest on ubuntu 14.04.4, still testing it).

nuno-or, 2016-04-14:
This is what worked for me, I believe the stereolabs sdk needs to be fixed to account for non-empty description field when ioctl with VIDIOC_ENUM_FMT gets called.
```
diff --git a/drivers/media/v4l2-core/v4l2-ioctl.c b/drivers/media/v4l2-core/v4l2-ioctl.c
index 24dc45f..661f5d2 100644
--- a/drivers/media/v4l2-core/v4l2-ioctl.c
+++ b/drivers/media/v4l2-core/v4l2-ioctl.c
@@ -1322,8 +1322,8 @@ static int v4l_enum_fmt(const struct v4l2_ioctl_ops *ops,
 		ret = ops->vidioc_enum_fmt_sdr_cap(file, fh, arg);
 		break;
 	}
-	if (ret == 0)
-		v4l_fill_fmtdesc(p);
+	/* if (ret == 0) */
+	/* 	v4l_fill_fmtdesc(p); */
 	return ret;
 }
```

atomoclast, 2016-04-15:
@nuno-or Where is that edit required? 

In the ZED_SDK_LINUX_x86_64_v______.run?

eduardohenriquearnold, 2016-04-20:
Is there a solution that does not require patching the kernel? Is there a plan to release a new version of the SDK that takes into account the v4l2 modification?

sl-braun, 2016-04-21:
Hi,

We have solved that for Kernel 4.2 ( more generally 4.X) in new release 0.9.4. It is currently in QA then should be released soon.

Best,
Olivier

xbklairith, 2016-04-22:
@sl-braun  i will look forward to 0.9.4
@nuno-or Thank you very much for your work.

adujardin, 2016-05-12:
Hi, 
Just to let you guys know that the [0.9.4](https://www.stereolabs.com/developers/#download_anchor) is out !

MargaritaDmitrienko, 2016-05-14:
Hello! I am experiencing a similar issue on Ubuntu 14.04.4 Kernel 3.10 Cuda 7.0. `/usr/local/zed/settings/` contains calibration file and `lsusb | grep 2b03:f580 | wc -l` gives 1. But Depth Viewer says "No ZED detected : Only ofline mode is available (with SVO files)". I can see following in terminal when run ZED Explorer:
`Cannot identify '/dev/video-1': 2, No such file or directory`
`*** buffer overflow detected ***: ./ZED Explorer terminated`
`Aborted`
I have no "video-1" file and also no "video0" file as @amiltonwong has.  What might be the problem?

sl-braun, 2016-05-16:
Hi,

The new kernel 4.X of Ubuntu 14.04.4 was not supported until v0.9.4 of the ZED SDK. Make sure you have downloaded it on the website (currently available).
It should solved your issue.

Best,
Olivier

MargaritaDmitrienko, 2016-05-16:
@sl-braun but I have an old Kernel 3.10. And 3 days ago everything was working fine but now does not. Probably someone (several people are using this computer) changed something but I don't know what. Should I update Kernel and Cuda?

atomoclast, 2016-05-26:
Is there an alternative to solving this aside from upgrading with CUDA 7.5? My system has other applications that need 6.5, so it would be easier to stick with that for now.

Thanks!
~Andrew

willdzeng, 2016-06-20:
I have the same issue, Zed is working for my USB2.0 but not 3.0.
On my another mother board, I turned on the "initialize USB3.0 for XHCI" and it works for USB3.0 also.
I guess you can try play with the BIOS usb configuration to see if that's the problem or not.

aorait, 2016-07-07:
I installed Jetpack 2.2 on TX1 and now my ZED camera is not detected. Any ideas on how to fix this?

Does anyone have the ZED SDK 0.9.4 version? Please let me know if you do.

Myzhar, 2016-07-08:
@aorait what kind of error are you getting?

Can you please cut and paste the result od "lsusb" command with your ZED connected?

aorait, 2016-07-08:
@Myzhar Below is the output of the lsusb command with ZED connected

Bus 001 Device 005: ID 1ec9:0081  
Bus 001 Device 004: ID 03f0:134a Hewlett-Packard 
Bus 001 Device 003: ID 413c:2107 Dell Computer Corp. 
Bus 001 Device 002: ID 2109:2811  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0955:09ff NVidia Corp. 
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The error is that when I run any of the samples or use the tools it gives an error "ZED not detected".
I even tried to use it in Windows. But it doesn't show up in the Explorer.

It was working fine with the 0.9.4 SDK and the previous JetPack.

Myzhar, 2016-07-15:
@aorait I heard about this error on Jetson TX1 from other users using the new SDK v1.0.0. Try to contact the support to have more information

adujardin, 2016-07-21:
Hi,

For now, you shouldn't use the Jetpack 2.2 with the ZED (but use the previous Jetpack 2.1), it came out after the release of the SDK 1.0 and we found that there are some issues with the USB3 communication with the ZED. Sorry for the inconvenience.

We are still investigating what causes the problem.

Razofiter, 2016-07-31:
Hello,

I am using a tk1 board(L4T 2.2.1) with zed(SDK 1.0).
I get this error when I run the file next file

**ubuntu@tegra-ubuntu:/usr/local/zed/tools$ ./ZED\ Explorer** 
ZED Explorer : ChangeResolution
ZED Explorer : changeFPS
ZED Explorer : changeFPS
ZED Explorer : Failed to setup device
Segmentation fault

Below I have a print screen obtained after running the Depth Viewer file

[image]

ALso, by running lsusb command I got this:
ubuntu@tegra-ubuntu:~/Documents/ZED$ lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 2b03:f580  
Bus 001 Device 009: ID 045e:07b2 Microsoft Corp. 
Bus 001 Device 007: ID 0b38:0003 Gear Head Keyboard
Bus 001 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

What do you think? Problem with version?
Thank you!

closed 2016-11-02 by adujardin

afwaan, 2017-12-12:
I am seeing similar issue. My camera starts but after sometime it throws "VIDIOC_G_PARM - Unable to get timeperframe: Bad file descriptor" 
and crashes. Any suggestions?

wujiang, 2017-12-19:
Same problem here. I got `VIDIOC_G_PARM - Unable to get timeperframe: No such device` sometimes. `lsusb` shows the device was still connected when this happened. Sometimes I got `Error code: Low USB Bandwidth` but by restarting it without doing anything else it worked again.

afwaan, 2017-12-19:
Yeah, similar happened to me. Restarted and all of a sudden it started working.

wujiang, 2017-12-19:
Opened a new issue #172.