stereolabs/zed-ros-wrapper #29: catkin_make error (ubuntu 14.04, ROS indigo)

amiltonwong, 2016-03-14:
Dear all,

I encounter the following catkin_make error:
Is it required to install ZED drive first in my system?
(My system ubuntu 14.04, ROS indigo)
====================================================

-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper)
Debug compilation active
CMake Error at zed-ros-wrapper/CMakeLists.txt:20 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 0.9) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/root/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/root/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
root@milton-OptiPlex-9010:~/catkin_ws# 

====================================================

adujardin, 2016-03-14:
Hi,

Yes you need to have the ZED SDK installed before trying to compile/use the samples

amiltonwong, 2016-03-15:
Hi, @adujardin ,

Thx, so I installed the ZED SDK . And then catkin_make the zed-ros-wrapper successfully. When I run the command: roslaunch zed_wrapper zed.launch
It prompts me :  [ INFO] [1458027143.090545829]: ZED_NOT_AVAILABLE
But I'm sure the ZED stereo camera is already plugged it usb port.

Any suggestion to fix this issue?

(Detailed log is listed as follows:)

===========================================================

milton@milton-OptiPlex-9010:~/catkin_ws$ roslaunch zed_wrapper zed.launch
... logging to /home/milton/.ros/log/0b3ac2fa-ea80-11e5-a652-90b11c63b99a/roslaunch-milton-OptiPlex-9010-19898.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://milton-OptiPlex-9010:34378/

SUMMARY
========

PARAMETERS
 * /camera/zed_wrapper_node/cloud_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /camera/zed_wrapper_node/depth_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/depth_topic: depth/image_rect_...
 * /camera/zed_wrapper_node/frame_rate: 30
 * /camera/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /camera/zed_wrapper_node/left_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/left_topic: left/image_rect_c...
 * /camera/zed_wrapper_node/point_cloud_topic: point_cloud/cloud
 * /camera/zed_wrapper_node/quality: 2
 * /camera/zed_wrapper_node/resolution: 2
 * /camera/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /camera/zed_wrapper_node/rgb_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /camera/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /camera/zed_wrapper_node/right_frame_id: /zed_optical_frame
 * /camera/zed_wrapper_node/right_topic: right/image_rect_...
 * /camera/zed_wrapper_node/sensing_mode: 1
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)
  /
    zed_base_link1 (tf/static_transform_publisher)
    zed_base_link2 (tf/static_transform_publisher)

auto-starting new master
process[master]: started with pid [19910]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0b3ac2fa-ea80-11e5-a652-90b11c63b99a
process[rosout-1]: started with pid [19923]
started core service [/rosout]
process[zed_base_link1-2]: started with pid [19937]
process[zed_base_link2-3]: started with pid [19941]
process[camera/zed_wrapper_node-4]: started with pid [19942]
[ INFO] [1458027138.807690493]: ZED_WRAPPER Node initialized
[ INFO] [1458027138.834462500]: Using ZED Camera
modprobe: ERROR: could not insert 'nvidia_352': No such device
ZED SDK >> (Init) Best GPU Found : .)���, ID : -1 
ZED SDK >> (Init) Quality mode has been set to Quality mode
[ INFO] [1458027139.088886596]: ZED_NOT_AVAILABLE
ZED SDK >> (Init) Best GPU Found : ���*, ID : -1 
ZED SDK >> (Init) Quality mode has been set to Quality mode
[ INFO] [1458027141.089728884]: ZED_NOT_AVAILABLE
ZED SDK >> (Init) Best GPU Found : .)���, ID : -1 
ZED SDK >> (Init) Quality mode has been set to Quality mode
[ INFO] [1458027143.090545829]: ZED_NOT_AVAILABLE
ZED SDK >> (Init) Best GPU Found : .)���, ID : -1 

===========================================================

closed 2016-04-21 by adujardin