stereolabs/zed-ros-wrapper #31: ZED not Detected

atomoclast, 2016-04-14:
Hello, I was previously using the ZED ROS wrapper on a previous computer for a project. Since we are moving the application to a different computer, we need to migrate the code and applications over.

During the setup of the ZED, I keep experiencing issues with it being able to connect to the internet. it says that it can connect to the ZED, but that there was no internet connection (although I can browse through a browser and ping through terminal without any issues). I did try some of the following suggested code:

/usr/local/zed/tools/ZED\ Settings\ App -sn {YourZEDSerialNumber}

because I kept getting the same error when I tried to run:

/usr/local/zed/tools/ZED\ Settings\ App -factory and got the same error. That it can connect to the ZED, but not connect to the camera to validate the SN I think? 
I do have a .conf file in the settings folder. It just won't open up the Depth Viewer or any of the other tools.

I have no idea what's going on or why it was so painless when I installed it last.

Thanks in advance!

sl-braun, 2016-04-21:
Hi,

Please go to http://calib.stereolabs.com to manually download your calibration file. 
Since 0.9.3, ZED Settings App has been moved to ZED Explorer, therefore please update your ZED SDK version with the last one available on our website.

Best,
Olivier.

atomoclast, 2016-04-21:
Hello,

I saw the latest SDK, but it is not compatible with CUDA 6.5. I can't install it because we have many other software modules running and utilizing 6.5 right now.

atomoclast, 2016-04-21:
I did also follow your instructions, and went to calib.stereolabs.com, did as instructed on the page, downloaded the .conf file and placed it in the proper directory. I tried to run the ZED Depth Viewer, and it failed.

[image]

closed 2016-07-21 by adujardin