stereolabs/zed-ros-wrapper #32: Cant open ZED after 0.9.3 update.

Bardo91, 2016-04-16:
Hi, I run into a problem while updating to the newest version of the ZED SDK. I couldn't find a general issues place to post-it, and there is not information about this error in [Support web page](https://support.stereolabs.com/hc/). Is there any place to put this kind of issues?

Basically, I installed CUDA 7.5 for the newest version, and the ran the ZED installer. Every seems to be fine until the calibration. Then this message appears: `*** Download Factory Calibration ***
ZED Detected, but cannot open Camera. Please check that you don't have another camera connected...`. 

I use a laptop, so I thought it was related with the embedded camera. I disabled it temporary running `echo '3-1.3:1.0' | sudo tee /sys/bus/usb/drivers/uvcvideo/unbind`. But the error keept.

The config file was not downloaded too. I downloaded it manually using in this page http://calib.stereolabs.com/ and using the reference number of the camera, and tried to run the ZED Explorer (In the new version is suppose to be also de Setting App). But the application crashed before showing anything.

I couldn't find any reference of that message, did anybody encounter this problem too?

sl-braun, 2016-04-21:
Hi,

The 0.9.3 is not compatible with new Linux Kernel 4.X yet. We are currently solving this issue in 0.9.4 release (available in a few days).

Bardo91, 2016-04-21:
Ok, thanks. I´m closing the issue.

closed 2016-04-21 by Bardo91