stereolabs/zed-ros-wrapper #34: Impossible to open camera

jocacace, 2016-05-10:
Hi to all,

Yesterday I received my first ZED camera :)! I am trying to working with it. In Windwos everything work perfectly but now I am trying to use Zed applications for Linux.

Following the steps reported on the website to start to operate with this camera, I installed opencv 2.4 and Cuda 7.5. 
Sadly when I try to install the SDK I got this error (or I think that is an error):

ZED Detected, but cannot open Camera. Please check that you don't have another camera connected...
rm: cannot remove ‘/usr/local/zed/tools/ZED Settings App’: No such file or directory
ZED SDK installation complete !

As consequences, the /usr/local/zed/settings is empty. Following another issue post, I manually downloaded the configuration file  SN1010.conf. But, in any case, all ZED samples crashes with segmentation fault. In addition, I tried the ros wrapper package. Even this way crash:

[ INFO] [1462864745.884115015]: ZED_WRAPPER Node initialized
[ INFO] [1462864745.898097528]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GeForce GT 630 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to Quality mode
[camera/zed_wrapper_node-4] process has died 

Just to provide you more information: I have not the directory /usr/local/zed/tools/ZED\ Settings\ App (I found it in the other issue post).

Can you suggest me how to solve this problem?

Regards

Esam-Bustaty, 2016-05-11:
I have the same error, all samples and tools fail when the camera is connected, I have `Ubuntu 14.04.4` with `4.2.0-36-generic` kernel, `openCV 2.4` and `CUDA 7.5`, and the hardware is `i5 6600k` on a `z170a Krait Gaming` motherboard with `GTX 980 Ti` GPU.

on console the output is:
```
ZED Explorer : Failed to setup device
Segmentation fault (core dumped)
```
`ZED Settings App` seems cancelled in the latest SDK version and replaced with `/usr/local/zed/tools/ZED\ Explorer --dc`
running `/usr/local/zed/tools/ZED\ Explorer --dc` outputs
```
*** Download Factory Calibration ***
ZED Detected, but cannot open Camera. Please check that you don't have another camera connected...
```
`luvcview -d /dev/video0 -s 2560x720 -i 15` is opening the camera and working fine.

I copied `.conf` files from windows to settings folder.

`lsusb | grep 2b03:f580 | wc -l` says 1

`lsusb | grep "Linux Foundation 3.0 root hub" | wc -l` outputs 2

adujardin, 2016-05-12:
Hi,
It is related to the linux kernel. The ZED SDK 0.9.3 is not compatible with the kernel 4. The [current release](https://www.stereolabs.com/developers/#download_anchor) 0.9.4 ([release note)](https://www.stereolabs.com/developers/documentation/API/page4.html) fix the issue.

Esam-Bustaty, 2016-05-12:
thanks @adujardin, i'm glad 0.9.4 is finally out, it did fix the issue.

closed 2016-05-12 by adujardin