stereolabs/zed-ros-wrapper #36: rviz Fixed Frame should be zed_point_cloud, not zed_optical_frame

lewisanderson, 2016-05-19:
HI,

I believe this section of the readme is incorrect:

If you want to see the point cloud, launch rviz, select zed_optical_frame in Displays->Global Options->Fixed Frame->zed_optical_frame. Then click on 'add' (bottom left), select the 'By Topic' tab, select point_cloud->cloud->PointCloud2 and click 'OK'.

Instead, it should be:

If you want to see the point cloud, launch rviz, select zed_optical_frame in Displays->Global Options->Fixed Frame->**zed_point_cloud**. Then click on 'add' (bottom left), select the 'By Topic' tab, select point_cloud->cloud->PointCloud2 and click 'OK'.

This is tested on a recent install following all the instructions. If I follow the current Readme steps, I get an error in RViz saying:
Transform [sender=unknown_publisher]
For frame [/zed_point_cloud]: Fixed Frame [zed_opticalFrame] does not exist

However, if I use zed_point_cloud as the frame, I get no error and the point cloud shows up.

nesnes, 2016-05-20:
That's quite strange, normally both of _zed_point_cloud_ and _zed_optical_frame_ should work.
I took a look at your error message and made a search for "zed_opticalFrame" and "opticalFrame" in the whole repository without success. Did you try to write it instead of selecting it from the list in rviz?

Is this just a typo when you wrote the issue? -> Can you reproduce the error to check? (copy/paste)
If not, what are the versions of ROS, RVIZ, and the zed-ros-wrapper?

lewisanderson, 2016-05-21:
When I open the dropdown in rviz, nothing appears (besides "map"). So I have to manually type the fixed frame value (either zed_point_cloud or zed_optical_frame). 

I tried copying and pasting (zed_optical_frame), and still got same error.

Perhaps there is something weird in my installation.
lewis@lewis-ubuntu-laptop:~/catkin_ws$ rosrun rviz rviz
[ INFO] [1463794594.900587138]: **rviz version 1.11.14**
[ INFO] [1463794594.900684176]: compiled against Qt version 4.8.6
[ INFO] [1463794594.900697138]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1463794595.119316288]: Stereo is NOT SUPPORTED
[ INFO] [1463794595.119399420]: OpenGl version: 3 (GLSL 1.3).
**ZED SDK 0.9.4e**
**ROS Indigo**

zed-ros-wrapper was cloned from git two days ago (Wednesday May 18)

nesnes, 2016-05-21:
They will be in the list ONLY if the zed_wrapper is started, you should never add it by hand. So start the wrapper, launch rviz when the wrapper is ready and tell me if you still have an empty fixed Frame list :)

lewisanderson, 2016-05-23:
Looks like I actually dont get a dropdown at all when I click on the little dropdown arrow. I did launch rviz after ros wrapper, via "rosrun rviz rviz". I also noticed rviz is very slow (1-2fps, and typing in the text boxes has 1-2sec latency). Perhaps there is actually something wrong with rviz?

But still, when I manually type "zed_point_cloud" into Fixed Frame it works, and "zed_optical_frame" does not work. 

[image]

nesnes, 2016-05-24:
Thank you for the screen shot, it seems that you have a global status warning about the fixed frame, can you copy/paste it? It starts with "No tf data. Actual error...", it should be linked to the problem!

closed 2016-10-19 by adujardin