stereolabs/zed-ros-wrapper #37: ROS launch wrapper cannot initiate the ZED SDK

prarinya-ntu, 2016-05-24:
I am using Ubuntu 14.04 with Nvidia Quadro K1100M
I installed CUDA 7.5 with Nvidia driver 364

I can run Depth viewer and other ZED tools properly. But when I launched the ROS wrapper (last master branch), the process has been killed after Creating ZED GPU mem ...

> 
> core service [/rosout] found
> process[zed_base_link1-1]: started with pid [8969]
> process[zed_base_link2-2]: started with pid [8970]
> process[camera/zed_wrapper_node-3]: started with pid [8971]
> [ INFO] [1464060430.789109043]: ZED_WRAPPER Node initialized
> [ INFO] [1464060430.807173093]: Using ZED Camera
> ZED SDK >> (Init) Best GPU Found : Quadro K1100M , ID : 0
> ZED SDK >> (Init) Disparity mode has been set to Quality mode
> ZED SDK >> (Init) Creating ZED GPU mem...
> [camera/zed_wrapper_node-3] process has died [pid 8971, exit code -11, cmd /home/XXX/XXX/catkin_ws_3d/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/XXX/.ros/log/3876594a-215e-11e6-b9c4-a4c49451bf4e/camera-zed_wrapper_node-3.log].
> log file: /home/XXX/.ros/log/3876594a-215e-11e6-b9c4-a4c49451bf4e/camera-zed_wrapper_node-3*.log
> 

I checked the zed_wrapper_node.cpp code. The process was killed in the zed->init() function
        err = zed->init(static_cast<sl::zed::MODE> (quality), -1, true);

Do you have any suggest for us where to check ?

prarinya-ntu, 2016-05-30:
I have both OpenCV2.4 and 3.0 in my machine. It seem that the ZED sdk wrapper called some of my ROS components built by OpenCV3. So I just recompile my workspace with OpenCV2.4. Problem solved.

closed 2016-05-31 by adujardin

znculee, 2016-06-29:
I had the exactly same problem. Could you tell more detail about how to compile your workspace with OpenCV 2.4? (Not just catkin_make?) Thanks a lot!!!

prarinya-s, 2016-06-29:
@znculee I am not sure if you have the same problem as mine. First of all, you may need to check if your opencv is built with CUDA enabled or not.

In my case, I have "cvbridge" component that particularly built for OpenCV3.0. So I just remove that component and use default cvbridge.

znculee, 2016-06-29:
@prarinya-s I have solve it with compile wrapper package and vision_opencv package (https://github.com/ros-perception/vision_opencv) simultaneously. Thanks!

griz1112, 2017-07-20:
ROS has its own copy of opencv. Currently with kinetic its 3.2. When you source the setup,bash script it makes sure it runs that copy of opencv. Sets it up in /tmp during runtime only. Hard to spot it since it disappears as soon as you stop the app that is using it. I have once piece of code that requires 2.4 3.1.1 and 3.2 all cuda enabled including the one that is private to ROS. In order to use cuda you have to grab the source for the ROS package and recompile it with cuda enabled. Uninstall the package from ROS install from the deb you made of opencv with cuda for ROS then reinstall ros-kinetic-desktop or whatever version you previously installed. This is the only way you can get cuda enabled opencv in ROS that will work consistently.  If you try and path around it or fiddle with pkgconfig sooner or later it will bite you. ROS does this with several libraries so just because you built them with cuda enabled it doesn't mean they will be that when called by ROS. Really made a difference in the app I was building which is a monocular slam system.