stereolabs/zed-ros-wrapper #38: [ WARN] [1466283648.235687635]: Wait for a new image to proceed

willdzeng, 2016-06-18:
Hi,

I am a software engineer in Transcend Robotics.
I am trying to use ZED as our vision camera to do SLAM.

When I test algorithm like Rtabmap, zed_wapper_node keep sending warning of wait for new images.
Is this happening because of computation power? or maybe the image published being occupied in the rtabmap algorithm something like that?

Thanks.
Di Zeng

adujardin, 2016-06-20:
Hi,

You can safely ignore this.
This warning appears when the processing is faster than the framerate of the camera. It means that the SDK still doesn't have a new image to work with, so it waits for a [few ms](https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L412).

It is not a problem and this should be more a `ROS_DEBUG` than a warning. In a next update, it will probably be removed since it isn't so useful.

willdzeng, 2016-06-20:
Cool, Yeah, it doesn't affect other programs to run also.

closed 2016-06-20 by willdzeng