stereolabs/zed-ros-wrapper #40: Nvidia TK1 & ZED ROS :  Wait for a new image to proceed

won13y, 2016-06-21:
I am using ZED Stereo Camera with Nvidia TK1(DJI Manifold)

I was able to successfully run the following commands:

ubuntu@tegra-ubuntu:~/catkin_ws$ roslaunch zed_wrapper zed.launch

    process[camera/zed_wrapper_node-4]: started with pid [4558]
    [ INFO] [1466473776.466851120]: ZED_WRAPPER Node initialized
    [ INFO] [1466473776.520543168]: Using ZED Camera
    ZED SDK >> (Init) Best GPU Found : GK20A, ID : 0 
    ZED SDK >> (Init) Quality mode has been set to Quality mode
    ZED SDK >> (Init) Creating ZED GPU mem...
    ZED SDK >> (Init) Trying self calibration... 
    ZED SDK >> (Init) Done...
    [ INFO] [1466473781.749374004]: SUCCESS
    [ INFO] [1466473783.766500542]: Reconfigure confidence : 80
    [ INFO] [1466473784.003238084]: Advertized on topic rgb/image_rect_color
    [ INFO] [1466473784.007348780]: Advertized on topic left/image_rect_color
    [ INFO] [1466473784.010071021]: Advertized on topic right/image_rect_color
    [ INFO] [1466473784.012525504]: Advertized on topic depth/image_rect_color
    [ INFO] [1466473784.015019822]: Advertized on topic point_cloud/cloud
    [ INFO] [1466473784.016921541]: Advertized on topic rgb/camera_info
    [ INFO] [1466473784.018865177]: Advertized on topic left/camera_info
    [ INFO] [1466473784.020766646]: Advertized on topic right/camera_info
    [ INFO] [1466473784.022737616]: Advertized on topic depth/camera_info

But the problem is that

When I run rviz in order to see the point cloud, launch rviz, select zed_optical_frame in Displays->Global Options->Fixed Frame->zed_optical_frame. Then click on 'add' (bottom left), select the 'By Topic' tab, select point_cloud->cloud->PointCloud2 and click 'OK'.

Then following messages occurs continuously, and the  ZED was reinitialized 

    [WARN] [1466473700.636144984]: Wait for a new image to proceed
    [ WARN] [1466473700.638489631]: Wait for a new image to proceed
    [ WARN] [1466473700.640739776]: Wait for a new image to proceed
    [ WARN] [1466473700.643201010]: Wait for a new image to proceed
    [ INFO] [1466473701.058440423]: Using ZED Camera
    [ INFO] [1466473701.058901936]: Reinit camera
    ZED SDK >> (Init) Best GPU Found : GK20A, ID : 0 
    ZED SDK >> (Init) Quality mode has been set to Quality mode
    ZED SDK >> (Init) Creating ZED GPU mem...

willdzeng, 2016-06-22:
#38

closed 2016-10-19 by adujardin

BRAQ, 2017-04-14:
Hello there,

Were you able to configure the ZED sensor with DJI Manifold successfully ? I have the Jetson TX1 and I want to use it with the DJI guidance kit ? Any suggestions or thoughts ? 

Thank you,