stereolabs/zed-ros-wrapper #41: Odom Topic and visual odometry

willdzeng, 2016-06-22:
Hi,
What algorithm is used in tracking the camera?
Can you add a parameters to disable this function just in case we have our own visual odometry.

sl-braun, 2016-06-23:
Hi,

It is our own SLAM algorithm.
if you don't call enableTracking() during Init() and getPosition() in your loop , then the tracking will not run.

adujardin, 2016-06-23:
Hi,

Just to clarify, if you don't subscribe to the odometry topic, `enableTracking` won't be called and the tracking will be disabled.

willdzeng, 2016-06-24:
Got you ~ What about TF? how to cancel the tf publisher?

If it's a SLAM, then how can I see the map?

adujardin, 2016-06-24:
To cancel the TF publishing simply comment/remove [this line](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed.launch#L3) from the launch file.

The map from the tracking is not accessible by the public API.

willdzeng, 2016-06-24:
Comment out the tf_publish won't cancel the tf publishing since in the source code
publishTrackedFrame()
is called every time in the runloop()

Suggest add a parameter: track_camera, so that whenever user want to use the visual odometry published by ZED they can use it. If not, they can save some computation power.

closed 2016-07-21 by adujardin