stereolabs/zed-ros-wrapper #42: Support for Ubuntu 16.04 and ROS kinetic

vincentrou, 2016-06-23:
It would be nice to have the ZED SDK on ubuntu 16.04 since it is the new LTS release.
I installed the ZED SDK 1.0 and try to compile the zed_wrapper but I get link errors :
/home/romea/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_node.cpp:388: undefined reference to `sl::zed::Camera::Camera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/romea/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_node.cpp:499: undefined reference to `sl::zed::Camera::enableTracking(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/romea/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_node.cpp:524: undefined reference to `sl::zed::Camera::Camera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/home/romea/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_node.cpp:544: undefined reference to `sl::zed::Camera::enableTracking(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
collect2: error: ld returned 1 exit status

But the cmake variable ${ZED_LIBRARIES} seems well set :
/usr/local/zed/lib/libsl_zed.so/usr/local/zed/lib/libsl_depthcore.so/usr/local/zed/lib/libsl_calibration.so/usr/local/zed/lib/libsl_tracking.so/usr/local/zed/lib/libsl_svorw.so/usr/local/zed/lib/libcudpp.so/usr/local/zed/lib/libcudpp_hash.so

adujardin, 2016-06-23:
Hi,

Yes, we're working on it. But we can't release it before CUDA 8 official release (the first compatible with Ubuntu 16.04). 

However, I don't think this error is related to Ubuntu 16. The libsl_zed used probably isn't the 1.0. You can check by calling this :
`nm -g /usr/local/zed/lib/libsl_zed.so | grep enableTracking`

If something comes out, it means that the linker doesn't take this one.

vincentrou, 2016-06-23:
I have install the 1.0 sdk on a fresh Ubuntu16.04 with cuda 7.5 and opencv 3.1
```
$ nm -g /usr/local/zed/lib/libsl_zed.so | grep enableTracking
000000000001bd10 T _ZN2sl3zed6Camera14enableTrackingERN5Eigen6MatrixIfLi4ELi4ELi0ELi4ELi4EEEbSs
```
It find the library /usr/local/zed/lib/libsl_zed.so because if I remove it from the CMakeLists.txt, I have a lot more link error.
It is the same for zed-opencv repo.

ruilimit, 2016-08-03:
I am trying to use ZED on Ubuntu 16.04 and had a similar issue with the zed_wrapper_node. When I ran catkin_make, I had the following error. Have you figured out how to run ZED successfully on Ubuntu 16.04? Thanks! 

`Base path: /home/rui/catkin_ws
Source space: /home/rui/catkin_ws/src
Build space: /home/rui/catkin_ws/build
Devel space: /home/rui/catkin_ws/devel
Install space: /home/rui/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/rui/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/rui/catkin_ws/build"
####
[ 20%] Built target zed_wrapper_gencfg
[ 60%] Built target zed_cpu_ros
make[2]: stat: /usr/local/zed/lib/libsl_tracking.so: Too many levels of symbolic links
make[2]: *** No rule to make target '/usr/local/zed/lib/libsl_tracking.so', needed by '/home/rui/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node'.  Stop.
CMakeFiles/Makefile2:2175: recipe for target 'zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/all' failed
make[1]: *** [zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

adurandp, 2016-08-08:
Hello,
I use the same configuration (Ubuntu 16.04, Cuda 7.5 and openCV 3.1) and I cannot compile the ROS wrapper.

CMakeFiles/zed_wrapper_node.dir/src/zed_wrapper_node.cpp.o: In function `main':
zed_wrapper_node.cpp:(.text.startup+0x2350): undefined reference to `sl::zed::Camera::enableTracking(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'

By typing nm -g ..., I obtain the results as the previous post.

Has anyone figured out how to deal with that issue?
Thanks

honeyasong, 2016-08-09:
how did you solve this problem,my friend.@vincentrou

honeyasong, 2016-08-09:
I met the same question as you,how did you solve this problem,my friend.@vincentrou

sl-braun, 2016-08-09:
Hi,

Current ZED SDK is not compatible with 16.04. We will provide the beta support along with CUDA 8 in the next release.

Best,

honeyasong, 2016-08-09:
ok.waiting for your best news.thanks.@sl-braun

bdholt1, 2016-10-10:
Hi, 

I'm at the same point as many of the previous posters here (Ubuntu 16.04, OpenCV 3.1) but just today I've installed CUDA 8 (https://developer.nvidia.com/cuda-toolkit) as well.

$ cd /usr/local/zed/sample/cpu/Simple\ Tracking/
$ mkdir build && cd build
$ cmake ..
$ make VERBOSE=1 -j8

/usr/bin/ld: warning: libcudart.so.7.5, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libsl_tracking.so, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnppc.so.7.5, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnppi.so.7.5, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libnpps.so.7.5, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)
CMakeFiles/ZED_Simple_Tracking.dir/src/main.o: In function `main':
main.cpp:(.text.startup+0x6f): undefined reference to `sl::zed::Camera::Camera(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
main.cpp:(.text.startup+0x1f6): undefined reference to `sl::zed::Camera::enableTracking(Eigen::Matrix<float, 4, 4, 0, 4, 4>&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'

I think the issues with enableTracking are because libsl_tracking.so is a symlink to itself:
brian@ws-1156:/usr/local/zed/lib$ ls -la
total 161884
drwx------ 2 brian brian      4096 Oct  7 14:47 .
drwx------ 9 brian root       4096 Oct 10 12:55 ..
-rwx------ 1 brian brian    233568 Oct  7 14:47 libcudpp_hash.so
-rwx------ 1 brian brian 158466064 Oct  7 14:47 libcudpp.so
-rwx------ 1 brian brian   1018336 Oct  7 14:47 libsl_calibration.so
-rw------- 1 brian brian   4360624 Oct  7 14:47 libsl_depthcore.so
-rwx------ 1 brian brian     52000 Oct  7 14:47 libsl_svorw.so
lrwxrwxrwx 1 brian root         17 Oct  7 14:46 libsl_tracking.so -> libsl_tracking.so
-rwx------ 1 brian brian   1615015 Oct  7 14:47 libsl_zed.so

Any ideas when a new library will be released?

nesnes, 2016-10-10:
Hi,
The 1.1.1 version of the ZED SDK is currently available on the website and bring the CUDA 8 compatibility.
Please update to this version and give me a feedback.
https://www.stereolabs.com/developers/#download_anchor

bdholt1, 2016-10-10:
That was a massive help - I hadn't seen the new release of the ZED SDK.  It's compiled and linked and seems to be working.

closed 2016-10-10 by nesnes

honeyasong, 2016-10-25:
@bdholt1,Hi I met the same problem as you,now I solved,it's very simple,just uninstall the zed you have installed  before and delete the folder “zed" and then reinstall it.