stereolabs/zed-ros-wrapper #43: How to run zed_wrapper in ros indigo  with OCV3.1 dependencies?

xepost, 2016-06-26:
Hi ,
I have Ros Indigo, Ubuntu 14.04, recently updated to ZED driver version SDK 1.0b which requires Opencv 3.1 for dependencies.  I have managed to compile  by  creating symbolic links with a OCV3.1 build  which is located in a non-standard directory, The main issue  when it runs it  requires cv_bridge , which is compiled with 2.4.8.   I get the following error after running zed.launch :

ZED SDK >> (Init) Best GPU Found : GeForce GTX 660M , ID : 0
ZED SDK >> (Init) Disparity mode has been set to QUALITY
ZED SDK >> (Init) Creating ZED GPU mem...
[camera/zed_wrapper_node-3] process has died [pid 8586, exit code -11, cmd /home/xepost/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_ome/xepost/.ros/log/0d91d35e-3ba2-11e6-a3fe-8c89a5070cef/camera-zed_wrapper_node-3.log].
log file: /home/xepost/.ros/log/0d91d35e-3ba2-11e6-a3fe-8c89a5070cef/camera-zed_wrapper_node-3*.log

Any suggestions how to make this package work?

willdzeng, 2016-06-26:
build the new cv_bridge of kinect_devel branch from source should fix this
problem.

xepost, 2016-06-27:
Hi Zeng,

I have  removed my OCV 2.4.12 and install OCV 3.1. As your suggestion downloaded  vision_opencv, put them in an empty src. Changed the name in CMakeLists and package.xml to cv3_bridge to avoid  further confusion, after initializing this new src I have compiled  with "cv_bridge"  and zed_ros_warpper.

The linking seems right: 
Linking CXX executable /home/xepost/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node
cd /home/xepost/catkin_ws/build/zed-ros-wrapper-master && /usr/bin/cmake -E cmake_link_script CMakeFiles/zed_wrapper_node.dir/link.txt --verbose=1
/usr/bin/c++     -Wno-deprecated -g     CMakeFiles/zed_wrapper_node.dir/src/zed_wrapper_node.cpp.o  -o /home/xepost/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node  -L/usr/local/cuda-7.5/lib64  -L/usr/local/zed/lib  -L/usr/local/lib -rdynamic /opt/ros/indigo/lib/libimage_transport.so -ltinyxml /opt/ros/indigo/lib/libclass_loader.so -lPocoFoundation -ldl /opt/ros/indigo/lib/libroslib.so /home/xepost/catkin_ws/devel/lib/libcv3_bridge.so /usr/local/lib/libopencv_xphoto.so.3.1.0 /usr/local/lib/libopencv_xobjdetect.so.3.1.0 /usr/local/lib/libopencv_ximgproc.so.3.1.0 /usr/local/lib/libopencv_xfeatures2d.so.3.1.0 /usr/local/lib/libopencv_tracking.so.3.1.0 /usr/local/lib/libopencv_text.so.3.1.0 /usr/local/lib/libopencv_surface_matching.so.3.1.0 /usr/local/lib/libopencv_structured_light.so.3.1.0 /usr/local/lib/libopencv_stereo.so.3.1.0 /usr/local/lib/libopencv_sfm.so.3.1.0 /usr/local/lib/libopencv_saliency.so.3.1.0 /usr/local/lib/libopencv_rgbd.so.3.1.0 /usr/local/lib/libopencv_reg.so.3.1.0 /usr/local/lib/libopencv_plot.so.3.1.0 /usr/local/lib/libopencv_optflow.so.3.1.0 /usr/local/lib/libopencv_line_descriptor.so.3.1.0 /usr/local/lib/libopencv_hdf.so.3.1.0 /usr/local/lib/libopencv_fuzzy.so.3.1.0 /usr/local/lib/libopencv_face.so.3.1.0 /usr/local/lib/libopencv_dpm.so.3.1.0 /usr/local/lib/libopencv_dnn.so.3.1.0 /usr/local/lib/libopencv_datasets.so.3.1.0 /usr/local/lib/libopencv_ccalib.so.3.1.0 /usr/local/lib/libopencv_bioinspired.so.3.1.0 /usr/local/lib/libopencv_bgsegm.so.3.1.0 /usr/local/lib/libopencv_aruco.so.3.1.0 /usr/local/lib/libopencv_viz.so.3.1.0 /usr/local/lib/libopencv_videostab.so.3.1.0 /usr/local/lib/libopencv_videoio.so.3.1.0 /usr/local/lib/libopencv_video.so.3.1.0 /usr/local/lib/libopencv_superres.so.3.1.0 /usr/local/lib/libopencv_stitching.so.3.1.0 /usr/local/lib/libopencv_shape.so.3.1.0 /usr/local/lib/libopencv_photo.so.3.1.0 /usr/local/lib/libopencv_objdetect.so.3.1.0 /usr/local/lib/libopencv_ml.so.3.1.0 /usr/local/lib/libopencv_imgproc.so.3.1.0 /usr/local/lib/libopencv_imgcodecs.so.3.1.0 /usr/local/lib/libopencv_highgui.so.3.1.0 /usr/local/lib/libopencv_flann.so.3.1.0 /usr/local/lib/libopencv_features2d.so.3.1.0 /usr/local/lib/libopencv_cudev.so.3.1.0 /usr/local/lib/libopencv_cudawarping.so.3.1.0 /usr/local/lib/libopencv_cudastereo.so.3.1.0 /usr/local/lib/libopencv_cudaoptflow.so.3.1.0 /usr/local/lib/libopencv_cudaobjdetect.so.3.1.0 /usr/local/lib/libopencv_cudalegacy.so.3.1.0 /usr/local/lib/libopencv_cudaimgproc.so.3.1.0 /usr/local/lib/libopencv_cudafilters.so.3.1.0 /usr/local/lib/libopencv_cudafeatures2d.so.3.1.0 /usr/local/lib/libopencv_cudacodec.so.3.1.0 /usr/local/lib/libopencv_cudabgsegm.so.3.1.0 /usr/local/lib/libopencv_cudaarithm.so.3.1.0 /usr/local/lib/libopencv_core.so.3.1.0 /usr/local/lib/libopencv_calib3d.so.3.1.0 /opt/ros/indigo/lib/libdynamic_reconfigure_config_init_mutex.so /opt/ros/indigo/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libxmlrpcpp.so /opt/ros/indigo/lib/libtf2.so /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge /usr/local/zed/lib/libsl_zed.so /usr/local/zed/lib/libsl_depthcore.so /usr/local/zed/lib/libsl_calibration.so /usr/local/zed/lib/libsl_tracking.so /usr/local/zed/lib/libsl_svorw.so /usr/local/zed/lib/libcudpp.so /usr/local/zed/lib/libcudpp_hash.so /usr/local/cuda-7.5/lib64/libcudart.so /usr/local/cuda-7.5/lib64/libnppi.so /usr/local/cuda-7.5/lib64/libnpps.so /usr/local/lib/libopencv_core.so.3.1.0 /usr/local/lib/libopencv_highgui.so.3.1.0 /usr/local/lib/libopencv_imgproc.so.3.1.0 /usr/local/lib/libboost_system.so /usr/local/lib/libboost_filesystem.so /usr/local/lib/libboost_thread.so /usr/local/lib/libboost_date_time.so /usr/local/lib/libboost_iostreams.so /usr/local/lib/libboost_serialization.so /usr/local/lib/libboost_chrono.so -lpthread /usr/local/lib/libpcl_common.so /usr/local/lib/libpcl_octree.so -lOpenNI -lOpenNI2 /usr/lib/libvtkCommon.so.5.8.0 /usr/lib/libvtkFiltering.so.5.8.0 /usr/lib/libvtkImaging.so.5.8.0 /usr/lib/libvtkGraphics.so.5.8.0 /usr/lib/libvtkGenericFiltering.so.5.8.0 /usr/lib/libvtkIO.so.5.8.0 /usr/lib/libvtkRendering.so.5.8.0 /usr/lib/libvtkVolumeRendering.so.5.8.0 /usr/lib/libvtkHybrid.so.5.8.0 /usr/lib/libvtkWidgets.so.5.8.0 /usr/lib/libvtkParallel.so.5.8.0 /usr/lib/libvtkInfovis.so.5.8.0 /usr/lib/libvtkGeovis.so.5.8.0 /usr/lib/libvtkViews.so.5.8.0 /usr/lib/libvtkCharts.so.5.8.0 /usr/local/lib/libpcl_io.so -Wl,-Bstatic -lflann_cpp_s -Wl,-Bdynamic /usr/local/lib/libpcl_kdtree.so /usr/local/lib/libpcl_search.so /usr/local/lib/libpcl_samp [... cut ...]
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) Done...
[ INFO] [1467027443.894571349]: SUCCESS
[ INFO] [1467027445.901814019]: Reconfigure confidence : 80
[camera/zed_wrapper_node-3] process has died [pid 11234, exit code -11, cmd /home/xepost/catkin_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/xepost/.ros/log/82c4a2be-3c5b-11e6-a87d-ac220b262275/camera-zed_wrapper_node-3.log].

Thanks for the help!!

willdzeng, 2016-06-28:
Hi,
you shouldn't rename it the cv_bridge to cv3_bridge, just build the whole vision_opencv in your workspace and source that workspace.

If you really want to rename the package, you should also change the package dependency in zed CMakeList.txt and package.xml, but I am not sure wheather that will cause issue.

Another bad thing about changing it is, other package may still try to use cv_bridge for opencv2 but you don't have it any more.

xepost, 2016-06-28:
Hi Zeng,

I have mentioned before but I already updated the  cv_bridge to cv3_bridge in the  cmake and packacage.xml. For not  having  ocv2 even if I install OCV3, libraries belonging to ros for ocv still remains(ocv2.4.8) inside the system. 

I have sent an email to zed support   lets see what they say...

willdzeng, 2016-06-28:
Hi,

I thought you only changed the CMakeList and package.xml in cv_bridge.

I think you should try build the whole vision_opencv package, instead of just cv_bridge. 
At least it works for me, maybe I am using jade and it's different for indigo.

nesnes, 2016-06-30:
Hi,
I had the same problem, with the exact same console output. In order to fix it, I took a look at the zed_wrapper code to find out how dependent it was from cv_bridge. I figured out that it was only used to convert the open_cv image to a ROS image message. So I took a breath, and I've extracted and adapted the only required method from the cv_bridge code to put it in the wrapper. 
Now, the zed_wrapper is no more dependent of cv_bridge and my pull request #48 has been merged.

You can just pull the repo and re-compile the wrapper with catkin_make. ;)

xepost, 2016-06-30:
Hi nesnes,

Thanks for the update, great work! One think to add the user need to edit  cmake line 32 to show the opencv3 build, for me it was:

find_package(OpenCV ${OCV_VERSION} COMPONENTS core highgui imgproc REQUIRED HINTS /home/xepost/opencv-3.1/build) 

Also to make things run properly we  need to make sure that OCV3 libraries are found by ZED SDK,   following  lines can be used for that:

my_build_dir=/home/$USER/opencv-3.1/build # dont forget to change your build dir!!!!!
sudo ln -s $my_build_dir/lib/libopencv*  /usr/lib/x86_64-linux-gnu/

If you do not want to clutter your lib current  ZED SDK  use only the following libs:
libopencv_video.so.3.1
libopencv_videoio.so.3.1
libopencv_imgproc.so.3.1
libopencv_imgcodecs.so.3.1
libopencv_highgui.so.3.1
libopencv_features2d.so.3.1
libopencv_core.so.3.1
libopencv_calib3d.so.3.1

sonyccd, 2016-07-06:
Are you saying that his rapper no longer depends on ROS opencv ?

nesnes, 2016-07-06:
It does need openCV, but it doesn't need cv_bridge anymore. It'll use the openCV found by ``find_package(opencv . . .)`` in its cmake file.

sonyccd, 2016-07-06:
But can it use the one built standard with ROS Indigo ?  Not OpenCV 3.1?

nesnes, 2016-07-06:
The ZED SDK is using OpenCV 3.1, so the wrapper must use the same.

You have to install OpenCV 3.1 (required before installing the ZED SDK), and then 'catkin_make' the wrapper. It should work out-of-the-box.

sonyccd, 2016-07-06:
Strange that I never got an error on install no having OpenCV 3.1.  Also I would recommend branching to ROS Kinetic as it defaults to OpenCV 3 where as Indigo is standard to 2.4.  Just to keep to the ROS REP and portability .

closed 2016-07-21 by adujardin