stereolabs/zed-ros-wrapper #47: Compilation error for zed-ros-wrapper

syamprasadkr, 2016-06-29:
I have a Jetson TX1 with Sample file 23.2 on it. It has the associated OpenCV4tegra as well. It also carries CUDA 7.0. While I try to install zed-ros-wrapper, it throws the error: 
`    CMake Error at /usr/share/OpenCV/OpenCVConfig.cmake:251 (if):
    if given arguments:

    "VERSION_LESS" "5.5"

    Unknown arguments specified
    Call Stack (most recent call first):
    zed-ros-wrapper/CMakeLists.txt:32 (find_package)

    -- Configuring incomplete, errors occurred!`

How do I fix this? I tried editing OpenCVConfig.cmake and added double quotes to VERSION_LESS and it stopped giving me error. But then it gives another error: `Invoking "make -j4 -l4" failed`
I tried the following solution : [https://github.com/stereolabs/zed-ros-wrapper/issues/12](url)
But it didn't fix the issue. Any ideas how to fix this?

nznobody, 2016-07-07:
Hello, I have had the same issue and simply overrode the settings as so:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index ace3b93..3979b3c 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -23,14 +23,14 @@ find_package(ZED 1.0 REQUIRED)
 exec_program(uname ARGS -p OUTPUT_VARIABLE CMAKE_SYSTEM_NAME2)
 if ( CMAKE_SYSTEM_NAME2 MATCHES "aarch64" OR CMAKE_SYSTEM_NAME2 MATCHES "armv7l" )
     SET(OCV_VERSION "2.4")
-    SET(CUDA_VERSION "")
+   # SET(CUDA_VERSION "")
 else()
     SET(CUDA_VERSION "7.5")
     SET(OCV_VERSION "3.1")
 endif()
 
 find_package(OpenCV ${OCV_VERSION} COMPONENTS core highgui imgproc REQUIRED)
-find_package(CUDA ${CUDA_VERSION} REQUIRED)
+find_package(CUDA  REQUIRED)
 
 find_package(PCL REQUIRED)
```

Unsure if this change is related but I also had too:
```

diff --git a/package.xml b/package.xml
index a08d41e..e94faf0 100644
--- a/package.xml
+++ b/package.xml
@@ -19,6 +19,7 @@
   <build_depend>cv_bridge</build_depend>
   <build_depend>image_transport</build_depend>
   <build_depend>dynamic_reconfigure</build_depend>
+  <build_depend>pcl_conversions</build_depend>
   
   <run_depend>tf2_ros</run_depend>
   <run_depend>nav_msgs</run_depend>
```

adujardin, 2016-07-07:
Hi, 

Could you test with the last version, please ? 
I made some change according to @nznobody solution.

syamprasadkr, 2016-07-08:
Hello,
Thanks for the reply. I am not sure if manually editing will affect its working in future. Anyway, the CUDA was installed again and the catkin workspace was removed. A new workspace was created and initialised. That fixed it .

syamprasadkr, 2016-07-08:
I am still interested to know the results of the changes made by @nznobody.

closed 2016-07-20 by syamprasadkr