stereolabs/zed-ros-wrapper #55: Linux Version of the SDK and ZED-ROS-Wrapper: Cuda 6.5 and 7.5 mix

solbach, 2016-08-05:
Hi,
I installed the ZED SDK 1.0. It relies on Cuda 7.5 and will abort the installation if Cuda 7.5 is not installed. However, if I try to run, e.g. _ZED Explorer_ it complains that Cuda 6.5 is not installed. 

During installation of the SDK:
>ERROR : the installer failed to detect the CUDA version (CUDA 7.5 required)

During execution of installed tool:
>error while loading shared libraries: libcudart.so.6.5: cannot open shared object file: No such file or directory

This mix up continues in the _ZED-ROS-Wrapper_. Running catkin results in
>CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message): Could NOT find CUDA: Found unsuitable version "7.5", but required is exact version "6.5" (found /usr/local/cuda-7.5)

If _SET(CUDA_VERSION "7.5")_ is changed in the _CMakeLists.txt_ to _SET(CUDA_VERSION "6.5")_ it compiles but depth calculation results in a segmentation fault. Segmentation fault also occurs by running the tool _ZED Depth Viewer_. To reach this point I installed cuda 6.5 alongside with 7.5 and played around with the order how the paths are set in _PATH_ and _LD_LIBRARY_PATH_.

Anyway, please let me know if I missed something. Otherwise, please clear up the dependencies of the SDK and the wrapper.

sl-braun, 2016-08-08:
Hi,

ZED Explorer is not linked against CUDA (6.5 or 7.5), but a link with OpenCV is made. Therefore can you check that OpenCV is either linked with CUDA 7.5 or not linked with CUDA at all (if you don't need gpuMat that's better).
Only keep one CUDA version (7.5) to avoid mixing conflicts.

Best,
/*OB*/

solbach, 2016-08-08:
OK. I removed all my OpenCV libs and reinstalled them by having only one CUDA 7.5 installed.

closed 2016-08-08 by solbach