stereolabs/zed-ros-wrapper #56: >> Detected S/N for ZED Camera : -1

harryman1, 2016-08-07:
I'm trying to install th ZED SDK in a Nvdia TX1 and I hav this message 

 >> Detected S/N for ZED Camera : -1

USB is setup to be 3.0 and I using the most recent version of the ZDK, 

please help

harryman1, 2016-08-08:
> Hi,
>
> On TX1, use the JetPack 2.1 instead of JetPack 2.2
>
> Best,
> /OB/
>
> -

Hi Oliver

Thank you, I just did that and the ZED Explorer has the same problem a segfault

I'm using Jetpack 2.1 32 bits on TX1 , and using the ZED SDK 1.0

What can I do to make it work ?

Thank you

sl-braun, 2016-08-09:
Hi. 
OK. Please open a support ticket (send a mail to support) so that we can make a full diagnostic of the camera.
Best,

closed 2016-10-19 by adujardin