stereolabs/zed-ros-wrapper #59: fatal error: Eigen/Core: No such file or directory

kdkalvik, 2016-09-04:
In file included from /home/ubuntu/catkin_ws/src/zed-ros-wrapper/src/zed_wrapper_node.cpp:54:0:
/usr/local/zed/include/zed/Camera.hpp:73:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/src/zed_wrapper_node.cpp.o] Error 1
make[1]: *** [zed-ros-wrapper/CMakeFiles/zed_wrapper_node.dir/all] Error 2
make: *** [all] Error 2

adujardin, 2016-09-05:
Hi, 
You seem to have an outdated version of the zed-ros-wrapper. The version you have was made for ZED SDK 0.9.X and you have the ZED SDK 1.X installed.

Make sure you download the latest version from the [master branch](https://github.com/stereolabs/zed-ros-wrapper/archive/master.zip).

kdkalvik, 2016-09-05:
My bad I tried the latest version and I get the following error 

CMake Error at zed-ros-wrapper/CMakeLists.txt:10 (find_package):
  Could not find a configuration file for package "ZED" that is compatible
  with requested version "1.1".

  The following configuration files were considered but not accepted:

    /usr/local/zed/zed-config.cmake, version: 1.0.0
    /usr/local/zed/zed-config.cmake, version: 1.0.0

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
ubuntu@tegra-ubuntu:~/catkin_ws$

kdkalvik, 2016-09-05:
I deleted the calibration file and reinstalled the latest ZED SDK and that did it. Thank you.

closed 2016-09-05 by kdkalvik