stereolabs/zed-ros-wrapper #60: URDF for ZED camera

ramamoorthyluxman, 2016-09-05:
I find the urdf for ZED camera is currently unavailable online. Could you please provide the same with the transformation between left, right cameras and the PCL? Also, it would be great if you can make the stl model of your camera available too. Thanks in advance.

nesnes, 2016-09-28:
Hi, sorry for the delay, I'll try to take a look at this. Do you have a sample file from wich it would be easy to start from?

reeve333, 2016-09-29:
Dear nesnes,

The following link is kinect urdf. You could use it as a reference.
https://github.com/turtlebot/turtlebot/blob/hydro/turtlebot_description/urdf/sensors/kinect.urdf.xacro

The following link is how to create urdf. 
http://wiki.ros.org/urdf/Tutorials
http://wiki.ros.org/tf/Tutorials

nesnes, 2016-10-10:
Hi,

I've committed the URDF support this morning. It contains the left right and center transform that you expected. 

By the way, you will notice that the origin of the point cloud values is the left camera.

Please give me a feedback,
Thanks for your help,
Best

closed 2016-10-10 by nesnes

ramamoorthyluxman, 2016-10-18:
Hi. I am sorry for the delayed response. 
Thank you very much for extending your help in providing the URDF.