stereolabs/zed-ros-wrapper #61: ZED with rtabmap not working

kdkalvik, 2016-09-05:
When I try to start rtabmap after selecting zed sdk as the camera the program crashes and gives me the following error
rtabmap: /usr/local/include/eigen3/Eigen/src/SVD/JacobiSVD.h:632: const MatrixVType& Eigen::JacobiSVD<MatrixType, QRPreconditioner>::matrixV() const [with _MatrixType = Eigen::Matrix<double, 3, 3>; int QRPreconditioner = 2; Eigen::JacobiSVD<MatrixType, QRPreconditioner>::MatrixVType = Eigen::Matrix<double, 3, 3>; typename _MatrixType::Scalar = double]: Assertion `computeV() && "This JacobiSVD decomposition didn't compute V. Did you ask for it?"' failed.
Aborted

kdkalvik, 2016-12-01:
I fixed the issue by installing eigen library. If you go to /usr/local/include/eigen3/Eigen/src/SVD/JacobiSVD.h you will probably see there is no JacobiSVD.h file.

brude-93, 2016-12-01:
I have an install of libeigen3-dev in /usr/include and the JacobiSVD.h is there. Did you install eigen from source then?

kdkalvik, 2016-12-01:
Yeah when I installed eigen from source it fixed the issue for me

>
> I have an install of libeigen3-dev in /usr/include and the JacobiSVD.h is
there. Did you install eigen from source then?

brude-93, 2016-12-01:
I install eigen from source and I'm still receiving the same error. I tried updating to the newest L4T and ZED SDK and that doesn't work either.

kdkalvik, 2016-12-01:
Do you have the header file at this location /usr/local/include/eigen3/Eigen/src/SVD/JacobiSVD.h?

brude-93, 2016-12-01:
Yes it's there.

kdkalvik, 2016-12-01:
Try using PCL 1.8

brude-93, 2016-12-07:
Hey again, so I've installed PCL 1.8 and Eigen3 from source successfully. Now when I run, I get a different error. "*** Error in `/home/ubuntu/Projects/tx1/build/tx1': double free or corruption (out): 0x0024c6a0 ***" Any idea where this might be coming from? It happens immediately after I try to initialize the ZED I see "ZED SDK >> (Init) Done..." before that error. Thanks!

kdkalvik, 2016-12-07:
Sorry I didn't have thus issue before. Please post the solution here if you do figure it out.

brude-93, 2016-12-08:
Hello again! I found the solution for me. Again, running an NVIDIA TX1 with L4T 23.2 Ubuntu 14.04, ROS Indigo, OpenCV4Tegra 2.4, ZED SDK 1.0.0. To get this to work I had to do the following:
1. Install Eigen3.2.10 from source
2. Install PCL-1.7.1 from source
3. Install RTABMAP 0.11.12 from source (without ROS, just standalone source install)

After I installed all these packages from source, I re-installed the ZED SDK and I now am able to run everything. The free() issue I was dealing with was an issue with freeing a pointer to a buffer where I was gathering point cloud data. I think we can close this thread now? Thanks again for the help!

tonepilot, 2017-02-06:
I have a NVIDIA TX1 also and ZED camera, and wish to install rtabmap. Are you aware of any other solutions since the above was posted, such as support for TX1 64-bit version?

closed 2018-06-28 by Myzhar