stereolabs/zed-ros-wrapper #62: Multiple ZED camera support

doisyg, 2016-09-11:
Hello,
This is more a question than an issue.
Is it possible to have multiple ZED cameras working with ROS on a single computer from a software perspective (assuming there are no hardware related issues like slow USB 3 controller)?
And if not, would a solution with multiple computers, ie one per camera, all on the same network would work?
Thanks!

nesnes, 2016-09-12:
Hi,
It is technically possible, but actually, you'll need to run a second zed-ros-wrapper with a different name, different topic names, and edit a bit the source code to select the second ZED instead of the first (by default). But if the same PC is computing the data of two ZEDs, it needs to be very powerful.

With ROS, you can subscribe to topics on other computers, so it should be possible to have several computers with one ZED on each, gathered by a master computer.

doisyg, 2016-09-12:
Great, that's excellent news. I will report here the results if we decide to go ahead with our project.

closed 2016-09-12 by doisyg