stereolabs/zed-ros-wrapper #67: The 'nan' and 'inf' values in depth image.

syamprasadkr, 2016-09-16:
I am trying to obtain depth value of every pixel in the image coming from a zed camera. I have the following approach. I have created a node based on [this](http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages). I have changed the topic to be subscribed to as` /camera/depth/image_rect_color `and the destination encoding as `32FC1`. 

I am running this on Jetson TX1 with ros indigo on ubuntu 14.04.

I have two issues:
1) When I display the depth image in OpenCV, I obtain a black and white image and not a grayscale image(to indicate different depths). What am I doing wrong?
2) I have the following block of code to print the depth values:
```
for (int i=0; i<cv_ptr->image.rows; i++)
    {
    	for (int j=0; j<cv_ptr->image.cols; j++)
        {
        	cv::Scalar intensity = cv_ptr->image.at<float>(i, j); 
        	double depth = intensity.val[0];
         	ROS_INFO("DEPTH INFO of pixel(%d,%d): [%lf]", i,j,depth);   
	}
     }
```
Here` cv_ptr->image` will give the OpenCV image. When I check the values, I have a lot of `nan` and `inf `values. I believe `inf ` indicates that the depth is out of the range of camera(20m). But why are some values 'nan'?

nesnes, 2016-09-16:
Hi,
Those values mean that the ZED SDK was not able to compute the depth on these pixels, probably because they are not visible from the two sensors of the ZED or because there is not enough characteristic elements to identify the pixel in the two images.
However, you can ask the ZED SDK to estimate the depth of these pixels by editing the launch file of the ZED ROS wrapper (zed-ros-wrapper/launch/zed.launch).
Just change the "sensing_mode" from "1" to "0". See the README.md file to have more information about these parameters.
OpenCV isn't great at displaying 32FC1 images, you should convert it to 8UC1 to have better results. (check your min and max value for your conversion)

syamprasadkr, 2016-09-16:
@nesnes 
Thank You for the reply. I am definitely interested in getting depth values directly from zed-ros-wrapper. I just need the depth value of all pixels to be available in ROS. I will definitely look into that.

But from a learning perspective, the ros messages which are being converted to OpenCV image are encoded in 32FC1. Do you think it is advisable to display the OpenCV image in 8UC1? Could this be the reason for black and white images?

nesnes, 2016-09-16:
Actually what openCV does to display your image is looking for the max and min value in the 32FC1 (32bit) image and creating a gray scale in 8UC1 (8bit) with it. If you have inf values, the max would be such high that the other values are barely displayed. If you take a closer look, you might be able to actually see something in your black image, but maybe not.

syamprasadkr, 2016-09-16:
@nesnes 
That makes a lot of sense, I will work on this and will let you know the results. Thank You.

syamprasadkr, 2016-09-16:
@nesnes 
I tried conversion to 8UC1. I did not get any 'nan' or 'inf'. But that gave me almost every value to be zero with some non-zero weird values. The values that I was getting when the destination format was 32FC1 made a lot of sense (They felt like depth values in meter). 
I also tried changing sensing mode from STANDARD to FILL in the zed-ros-wrapper. But it did not work.

nesnes, 2016-09-19:
Hi,
You may try to change the depthmap values from meter to millimeter to avoid floats.
Just change the line 81 of the wrapper to 1, and recompile by moving to the root of your catkin workspace and running "catkin_make".
https://github.com/stereolabs/zed-ros-wrapper/blob/master/src/zed_wrapper_node.cpp#L81

syamprasadkr, 2016-09-19:
I tried that already. But I have since problems accessing the pixel values in 16 bit mode. I am solving the issue. I shall update soon.

syamprasadkr, 2016-09-21:
I was trying the 16 bit mode. I changed destination encoding to 16UC1. But I am a bit confused about the following block of code:

```
for (int i=0; i<cv_ptr->image.rows; i++)
    {
        for (int j=0; j<cv_ptr->image.cols; j++)
        {
            cv::Scalar intensity = cv_ptr->image.at<float>(i, j); 
            double depth = intensity.val[0];
            ROS_INFO("DEPTH INFO of pixel(%d,%d): [%lf]", i,j,depth);   
    }
     }

```
In the line: `cv::Scalar intensity = cv_ptr->image.at<float>(i, j); ` what should replace `float`? 

I tried `unsigned __int16` and `__int16`. But that threw the error: `template argument 1 is invalid` during `catkin_make` . The `ushort` worked, but it rounded off the depth values to nearest integer (Instead of seeing values like 1.234, 3.456 etc, I was seeing 1.000 and 3.000)

closed 2018-06-28 by Myzhar

BunnyIcon, 2020-11-18:
Hello, I'm facing the same problem, did you solve it? There are still some zeros in my depth array.

Myzhar, 2020-11-18:
Hi @BunnyIcon 
please do not reply to a closed issue. Opening a new one is always preferred to better keep track of your questions