stereolabs/zed-ros-wrapper #68: Error Detected:  ZED_SETTINGS_FILE_NOT_AVAILABLE

shoemakerlevy9, 2016-09-21:
Hi,
I'm trying to use the ZED camera on a Jetson TX1 running Ubuntu 14.04. I installed the SDK for the TX1 and then manually installed my .config file. My serial number is 1962 so I downloaded SN1962.conf and placed it in my setting folder.  I then go to open up the 'ZED explorer' under tools and nothing shows up. I then open 'ZED Depth Viewer' and get the error: 

> Error Detected:  ZED_SETTINGS_FILE_NOT_AVAILABLE

How to I fix this issue seeing as I have my .conf file in the setting folder.

Thanks

nesnes, 2016-09-23:
Hi, if your jetson is running Jetpack 2.2 64bit, please install this specific release of the SDK : https://www.stereolabs.com/download_327af3/QA_Version/ 
Also check the permissions of the config file (and above folders), it needs to be readable and writable by any user.

augustjd, 2016-10-13:
I'm having the same issue - I've been able to track down that somehow, the ZED camera which is plugged into my system is reporting its SN as -1, literally, negative one. This causes lots of problems with various ZED components, because it winds up getting passed around on the command line, where it is interpreted as a flag and not a number.

I don't have a solution for you, but for me the problem is intermittant. Possibly some combination of restarting the machine and unplugging/replugging the camera may be able to solve it.

gmhuili, 2016-10-17:
I faced the same issue (ZED_SETTINGS_FILE_NOT_AVAILABLE) when I was installing it on Jetson TX1 running Ubuntu 14.04 32-bit. Both /dev/video0 and /dev/video1 are created. And the /usr/local/zed/settings is empty. When I tried to run "/usr/local/zed/tools/ZED\ Settings\ App -factory" to download the settings, I found all apps have been renamed and there is no settings app any longer.

adujardin, 2016-10-19:
@gmhuili 'ZED Settings App' and 'ZED Explorer' has been merged since the ZED SDK 0.9.3. You can get the calibration file by calling :
`/usr/local/zed/tools/ZED\ Explorer --download_calibration # (or --dc)`

You can also get it at https://calib.stereolabs.com 

Also if you haven't updated in a while, your ZED might need a firmware update. This step requires a desktop PC (Windows or Linux) since there are some issues with the Tegra (firmware corruption)

closed 2016-11-02 by adujardin