stereolabs/zed-ros-wrapper #70: ROS Support of Multi Input ZED Camera

reeve333, 2016-09-28:
Hi, I'm working on a robotics project with using 4 ZED Camera running simultaneously with ROS. 

I successful run all camera simultaneously. Unfortunately, I found that "Zed-ROS-Wrapper" is not support multi input yet. Therefore, Would I know is these feature will be available any time soon?

On the other hands, ZED Camera URDF is not available yet. Would ZED support team provide it soon?

Thank you very much. 

Best Regards,
Reeve

nesnes, 2016-09-28:
Hi,
That would be a nice feature in fact! Are you motivated to send us a pull request?

reeve333, 2016-09-29:
Hi nesnes, Yes, It will be nice features. I didn't wrote the features yet that's y I asked for Zed support team to provide if possible.

nesnes, 2016-09-29:
Hi,
I'll suggest this update to the team. The URDF file was also asked in issue #60, we're gonna look at it.

reeve333, 2016-09-30:
Thank you very much nesnes. Looking forward to it

nesnes, 2016-10-10:
Hi, I've just added the URDF support. I'll take a look to the multi camera support very soonly!

reeve333, 2016-10-11:
Thank you nesnes. Looking forward to the multi camera support as well.

adujardin, 2016-10-11:
Hi,
I developed a prototype version that supports multiple ZED and SVO files (at the same time). Check out the branch 'multi_zed' to get it. 
It probably not the optimum way of doing it. Feel free to give me feedback or even better submit a patch if something doesn't work as expected.

Also please note that the ZED take the full USB 3 bandwidth and take a significant amount of GPU memory, so you might have to reduce the framerate of the ZED (in case of green and purple frames, it's defined in the launch file) and/or depth quality.

adujardin, 2016-10-13:
I've also added new launch files for the [multi ZED support](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed_multi_cam.launch) (and [multi-GPU](https://github.com/stereolabs/zed-ros-wrapper/blob/master/launch/zed_multi_gpu.launch)). You can define every parameter for each ZED (nothing is done automatically, unlike the branch multi_zed). Each ZED is then handled by a separate process. Please note that the topics also has to be renamed manually in the launch file.

I think this method is better, it gives a greater control over what's happening.

closed 2016-10-13 by adujardin

reeve333, 2016-12-12:
Hi,

Could I know is that possible to set static ID to the Zed Camera?

My scenario is I have 4 Zed Camera launching at the same time, but I couldn't stabilize it (Camera switching all the time) and couldn't set fix the camera topic. Could I get suggestion how to fix it or request for the feature?

Meanwhile, I have update some coding for multi-camera launching. Should I make a proper pull request to the git?

Thank you very much.