stereolabs/zed-ros-wrapper #73: Poor pointcloud, adjust depth clamp setting

sjdrc, 2016-10-03:
I'm getting a very bad output of the point cloud from the ZED camera on the Jetson TX1. I get a similar output from the ZED Depth Viewer until I adjust the 'Depth Clamp' setting from the gear icon. How can I change this setting in the node? I've attempted to add this myself referring to Camera.setDepthClampValue() but anything I tried only gave me errors.

Can someone please give me some pointers?

nesnes, 2016-10-03:
Hi,
Can you explain a bit more what kind of bad output you're getting?

To introduce a fixed depth clamp in the wrapper, just add "zed->setDepthClampValue(5.0)" at line 541 of the wrapper. The values should be in meters. (if it doesn't work, try millimeters)

sjdrc, 2016-10-09:
Hey, sorry for the delayed response.

This is what I see with the stereolabs version of zed-ros-wrapper (fresh clone)
[image]

This is what I see with the kathoum fork with retreiveMeasure_gpu implemented (fresh clone)
[image]

In both versions, flat surfaces do not appear as flat in the point cloud.

My setup:
- Jetson TX1
- L4T 23.1
- CUDA 7.0
- PCL 1.8 (compiled from source with gpu enabled)
- OpenCV4Tegra
- ROS Indigo (compiled from source)

nesnes, 2016-10-10:
Hi,
Thanks for the clear and documented issue!
I'm currently working on this problem, you should have some news soonly.

nesnes, 2016-10-10:
Hi, I've just pushed a fix, can you confirm that the problem is solved?

sjdrc, 2016-10-11:
Hey, I've since done a fresh wipe and install of:
 - L4T 24.2
- CUDA 8.0
- PCL 1.7 (from repos)
- OpenCV4Tegra
- ROS Kinetic (from repos)

I'm still experiencing a similar issue. The cloud also flickers a lot now and random frames are all green and magenta. Furthermore, people wearing black jackets for some reason have invisible torsos.

[image]

I also had to run ```catkin_make -DCUDA_USE_STATIC_CUDA_RUNTIME=OFF``` in order to get it to build, otherwise it would complain about not being able to find -lopencv_dep_cudart.

There is also an error when running the wrapper node

```
ubuntu@tegra-ubuntu:~$ roslaunch zed_wrapper zed.launch
... logging to /home/ubuntu/.ros/log/98be14a8-8f6a-11e6-b1ad-00044b580a12/roslaunch-tegra-ubuntu-5582.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:54015/

SUMMARY
========

PARAMETERS
 * /camera/zed_wrapper_node/cloud_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /camera/zed_wrapper_node/depth_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/depth_topic: depth/image_rect_...
 * /camera/zed_wrapper_node/frame_rate: 60
 * /camera/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /camera/zed_wrapper_node/left_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/left_topic: left/image_rect_c...
 * /camera/zed_wrapper_node/odometry_DB: 
 * /camera/zed_wrapper_node/odometry_frame_id: /zed_initial_frame
 * /camera/zed_wrapper_node/odometry_topic: odom
 * /camera/zed_wrapper_node/odometry_transform_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/openni_depth_mode: 0
 * /camera/zed_wrapper_node/point_cloud_topic: point_cloud/cloud
 * /camera/zed_wrapper_node/quality: 1
 * /camera/zed_wrapper_node/resolution: 2
 * /camera/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /camera/zed_wrapper_node/rgb_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /camera/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /camera/zed_wrapper_node/right_frame_id: /zed_tracked_frame
 * /camera/zed_wrapper_node/right_topic: right/image_rect_...
 * /camera/zed_wrapper_node/sensing_mode: 1
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.5

NODES
  /camera/
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)
  /
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)

auto-starting new master
process[master]: started with pid [5593]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 98be14a8-8f6a-11e6-b1ad-00044b580a12
process[rosout-1]: started with pid [5606]
started core service [/rosout]
process[zed_base_link1-2]: started with pid [5623]
process[zed_base_link3-3]: started with pid [5624]
ERROR: cannot launch node of type [robot_state_publisher/state_publisher]: robot_state_publisher
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ubuntu/ani/src
ROS path [2]=/opt/ros/kinetic/share
process[camera/zed_wrapper_node-5]: started with pid [5630]
[ INFO] [1476159868.315374307]: ZED_WRAPPER Node initialized
[ INFO] [1476159868.368083227]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : NVIDIA Tegra X1 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) Done...
[ INFO] [1476159872.042415791]: SUCCESS
[ INFO] [1476159874.106146238]: Reconfigure confidence : 80
[ INFO] [1476159874.228371596]: Advertized on topic rgb/image_rect_color
[ INFO] [1476159874.231674630]: Advertized on topic left/image_rect_color
[ INFO] [1476159874.234030280]: Advertized on topic right/image_rect_color
[ INFO] [1476159874.236309108]: Advertized on topic depth/image_rect_color
[ INFO] [1476159874.238359657]: Advertized on topic point_cloud/cloud
[ INFO] [1476159874.240532445]: Advertized on topic rgb/camera_info
[ INFO] [1476159874.243304756]: Advertized on topic left/camera_info
[ INFO] [1476159874.245425930]: Advertized on topic right/camera_info
[ INFO] [1476159874.247613977]: Advertized on topic depth/camera_info
[ INFO] [1476159874.252094290]: Advertized on topic odom
```
This is resolved by installing ```ros-kinetic-robot-state-publisher```

I can also confirm that this issue occurs in the ZED Depth Viewer from the SDK on the Jetson TX1.

I'm currently setting up a windows machine to test the SDK with and I will set up a Ubuntu 16.04 desktop machine to test the SDK and node with to see if this is isolated to the Jetson.

Do you want me to test with L4T 23.1 again? Please let me know what I can do to help out :smile:

nesnes, 2016-10-11:
Hi,
It looks like you didn't get the last version of the wrapper. For example, the `-DCUDA_USE_STATIC_CUDA_RUNTIME=OFF` isn't needed in the last version.
https://github.com/stereolabs/zed-ros-wrapper/blob/master/CMakeLists.txt#L25

Please also make sure you're using the latest SDK : https://www.stereolabs.com/developers/#download_anchor

sjdrc, 2016-10-12:
Hey, I'm definitely using the latest version of the wrapper, I've done a fresh clone. I can see that CMakeLists has that line but when running ```catkin_make``` I still get the error ```/usr/bin/ld: cannot find -lopencv_dep_cudart```

When I run ```catkin_make -DCUDA_USE_STATIC_CUDA_RUNTIME``` then it compiles without an error. I am definitely using the 1.1.1 version of the SDK, as I have done a total fresh install on the Jetson TX1.

nznobody, 2016-10-12:
@sjdrc I am using the Jetson TX1 23.1 (or .2, can't remember), Ubuntu 14.04, 32b with Cuda 7.0 and the latest version works fine (see my closed issue). Also used SDK 1.1.0 there. Only thing I had to change was cuda version in CMakeLists file. Don't know if any of this helps, but my 2 cents.

sjdrc, 2016-10-12:
@nznobody Thanks for the info. Could you please do me a favour and pan down to show how flat a wall or floor surface appears in the ZED Depth Viewer? I think that my camera may be faulty.

nznobody, 2016-10-12:
I hope this is what you are looking for:

[image]

However look at my new issue #78 if you like.

sjdrc, 2016-10-12:
Thanks, that is very useful. I'll be tracking the progress of #78 closely.

closed 2018-06-28 by Myzhar

kartikpaigwar, 2019-07-01:
Hey, I am too getting a very bad point cloud. A flat surface (floor in my case) seems to get curvy. And the whole point cloud looks pretty bad. Does anyone have the same issue as me and found any ways to correct it. Please help. I am using Zed Mini, latest SDK, Cuda 10.
[image]
[image]
[image]