stereolabs/zed-ros-wrapper #74: Minimum depth clamp

nznobody, 2016-10-06:
[image]
[image]

Software versions: ZED SDK 1.1.0 ARM (Jetson TX1)
Current master of this repo (at date of writing)

Previous versions: ZED SDK 1.0.0 ARM (Jetson TX1)
Commit SHA: #05e47ffe5519b449c68d0032ce9995839ebd637e

Issue:

- Previously the camera worked fine and grabbed a full point cloud. 

- Now it appears to be clipping close points and not publishing them. See my launch file at the end for a reference. Not sure if this issue is with the SDK or the wrapper, but the Depth Explorer (First image) appears to show close points (Both point clouds have camera in same location). 

- The second image in in RVIZ using default / and modified code from latest master (doesn't change min clamp. See in the second photo the selected point at roughly 2.9m distance from the sensor).

Resolutions tried:

1.  sl::zed::InitParams.minimumDistance, have tried setting this to 0.0, 0.1, 0.5 as per attached git patch file. No change. Setting to 5.0 does make all points disappear so it seems to be working.

2.  zed->setDepthClampValue(10.0); Again, tried various values. Setting it low (e.g 0.5) makes all points disappear, seems to be working as intended

3. confidence = 100; Forcing confidence to 100, or manually setting it. Again, setting lower values makes points disappear, but doesn't resolve issue.

Therefor I am at a loss, and would like to see if anyone else has encountered a similar issue?

**Appendix:**

Launch file
```
<launch>
  <include file="$(find zed_wrapper)/launch/zed_tf.launch" />
  <arg name="svo_file" default=""/>
  <group ns="camera">
    <node name="zed_wrapper_node" pkg="zed_wrapper" type="zed_wrapper_node" args="$(arg svo_file)" output="screen">
      <param name="resolution"            value="3" />
      <param name="quality"               value="3" />
      <param name="sensing_mode"          value="1" />
      <param name="frame_rate"            value="15" />
      <param name="odometry_DB"           value="" />
      <param name="openni_depth_mode"     value="0" />

      <param name="rgb_topic"             value="rgb/image_rect_color" />
      <param name="rgb_cam_info_topic"    value="rgb/camera_info" />
      <param name="rgb_frame_id"          value="/zed_tracked_frame" />

      <param name="left_topic"            value="left/image_rect_color" />
      <param name="left_cam_info_topic"   value="left/camera_info" />
      <param name="left_frame_id"         value="/zed_tracked_frame" />

      <param name="right_topic"           value="right/image_rect_color" />
      <param name="right_cam_info_topic"  value="right/camera_info" />
      <param name="right_frame_id"        value="/zed_tracked_frame" />

      <param name="depth_topic"           value="depth/image_rect_color" />
      <param name="depth_cam_info_topic"  value="depth/camera_info" />
      <param name="depth_frame_id"        value="/zed_tracked_frame" />

      <param name="point_cloud_topic"     value="point_cloud/cloud" />
      <param name="cloud_frame_id"        value="/zed_tracked_frame" />

      <param name="odometry_topic"                value="odom" />
      <param name="odometry_frame_id"             value="/zed_initial_frame" />
      <param name="odometry_transform_frame_id"   value="/zed_tracked_frame" />

    </node>
  </group>
</launch>
```

git diff file:

[zed_depth_issue.txt](https://github.com/stereolabs/zed-ros-wrapper/files/512719/zed_depth_issue.txt)

nznobody, 2016-10-06:
I just noticed during launch I get:

`WARNING : (Init) the specified minimum depth value is closer than the threhold, it is rejected.`

nesnes, 2016-10-06:
Hi,

Someone else already had this issue and resolved it by modifying the ROS wrapper to retrieve the GPUpoint cloud and using cudaMemCopy to bring it back to CPU.
We are actually looking into it. It could help if you can tell me if you have the same issue with an above version of the wrapper :
https://github.com/stereolabs/zed-ros-wrapper/tree/05e47ffe5519b449c68d0032ce9995839ebd637e

Thanks for your help.

nznobody, 2016-10-09:
Sorry for the delayed response;

I have tested the mentioned commit: No luck, see attached. I believe it is something that changed in the underlying API because: When I use the ZED Depth explorer I can see the pointcloud fine BUT when I try to save to file (PCD or XYZ), I only get saved what I would see in the wrapper (A narrow band of points). 

Additional thoughts: What PCL are you using? I am using a master build from their git:
Using **git show**:
```
commit d237b9441a85cb4645dbfb7f1474efc8c273474f
Merge: d31f6be 8d97988
Author: Sergey Alexandrov <alexandrov88@gmail.com>
Date:   Mon Jun 20 10:14:00 2016 +0200

    Merge pull request #1560 from UnaNancyOwen/fix_realsense

    Fix error C2065 in RealSense Grabber on MSVC
```
I would rather not try muck around with PCL as it used to work on this version and takes about 12 hours to build nativly on the jetson >.<

What build arguments are you passing to cmake / make? I use (For Jetson TX1, Ubuntu 32b 14.04):

`-DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_FLAGS=-std=gnu++11 -Ofast -march=armv8-a -mrestrict-it -mfpu=neon-fp-armv8 -fpermissive`

Again, I have always used these, never had issues before.

Looking forward to working around this as it is a little bit of a hold up, especially cause I cannot even simulate capture data by saving PCD files anymore!

Photos:
First again the Zed Depth Explorer, then again the basic ros_wrapper (using resolution 3, quality 3, framerate 15, although this doesn't change much). Again notice the narrow band of points only.
[image]
[image]

nesnes, 2016-10-10:
Hi, I've just committed a fix, can you confirm that you now get the full point cloud?

nznobody, 2016-10-10:
Thanks you @nesnes , I can confirm this now produces a full point cloud. I was going to try update to SDK 1.1.1, but it states only for ubuntu 16.04. Is this true? A lot of people use the jetson with 14.04 because ROS isn't easy on aarch64. Will mark this as solved, but maybe PM me? Thanks again.

[image]

closed 2016-10-10 by nznobody