stereolabs/zed-ros-wrapper #76: Installation script requirements

bnascimento, 2016-10-06:
Latest jetson Jetpack 2.3 installs cuda 8 and ubuntu 16.04. Current script is not compatible and installation fails. When can we expect an installation script that runs on the latest jetpack 2.3.

Best regards, 
Bruno

nesnes, 2016-10-06:
Hi,
When you have this kind of issue, please provide more information about the fail.
A quick look into the CMakeFile.txt pointed me to the line 16 where CUDA 7.0 is specified.
https://github.com/stereolabs/zed-ros-wrapper/blob/master/CMakeLists.txt#L16
Just change this to CUDA 8 and tell me if it works.

golovanevAA, 2016-10-06:
At me it has turned out to start in jetson Jetpack 2.3 for this purpose I has changed CMakeLists.txt
SET (OCV_VERSION "2.4.13")
SET (CUDA_VERSION "8.0")
set(CUDA_USE_STATIC_CUDA_RUNTIME OFF)

closed 2018-06-28 by Myzhar