stereolabs/zed-ros-wrapper #80: zed ros wrapper problem?

hala1994, 2016-10-17:
I've downloaded zed ros wrapper successfully following this site:
http://qiita.com/furushchev/items/ef6b0c1b471cef6228c0
but since I use JTK1 I skipped all before the ros section 

note: I also downloaded ros indigo desktop full before following the previous website

when I run the commands :
roslaunch zed_wrapper zed.launch in terminal tap and
roslaunch zed_rtabmap.launch in another tap 

the program start but it shows lines and it construct the 3d model very very slowly when I try to change some settings it suddenly closed giving me this error:

[camera/zed_wrapper_node-5] process has died [pid 7520, exit code -9, cmd /home/ubuntu/ros/indigo/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/c9104cc8-9444-11e6-850b-f4f26d1c8979/camera-zed_wrapper_node-5.log].
log file: /home/ubuntu/.ros/log/c9104cc8-9444-11e6-850b-f4f26d1c8979/camera-zed_wrapper_node-5*.log

when I tried to run the command:
rosrun rviz rviz

it opened the program showing the grid empty when I tried to add panel to add the /point_cloud/cloud/PointCloud2 so I can see the live mapping it suddenly closed giving me this error:

[ INFO] [1476694905.957255060]: rviz version 1.11.14
[ INFO] [1476694905.957656056]: compiled against Qt version 4.8.6
[ INFO] [1476694905.957802138]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1476694906.580238824]: Stereo is NOT SUPPORTED
[ INFO] [1476694906.580583821]: OpenGl version: 4.4 (GLSL 4.4).

I don't know what is wrong or what I am doing wrong can you please help me?

adujardin, 2016-10-17:
Hi,
I think the TK1 is not powerful enough for your need. The ZED SDK on the K1 already take a lot of resources, rtabmap is also quite demanding. The crash might be caused by the memory limitation.

To get an acceptable framerate for live applications, you'll need at least a TX1.

hala1994, 2016-10-17:
is there any way to overcome this problem aside from changing to JTX1?

hala1994, 2016-10-17:
all I need is to produce a high quality 3d model but the zedfu isn't available on Ubuntu so I asked the zed team and they give me 2 choices 
1.to record svo video on Ubuntu and export it to another thread on windows 
2.to download rtabmap 
they advised me that opencv algorithm is not very good for my task
so I downloaded rtabmap in the manner I mentioned above 
is there any tool to help me with my task with the available resources:zed camera ,nvidia Tegra K1,Ubuntu

adujardin, 2016-10-17:
You can try using rtabmap standalone (ROS tend to use some resources), and try tweaking the parameters to get the lightest process possible. Try disabling stuff like live visualization (if possible, I'm not very familiar with it), loop closure, and even installing a lighter desktop environment on the K1 (like lxde) which take much less memory than unity.
Also, you should work with an SVO file, you won't be able to get a sufficient framerate to get a reliable tracking otherwise.
It depends on what you intend by "high-quality 3D model" but keep in mind that stereo reconstructions can be noisy, especially for indoor use and with limited compute and memory capabilities.

hala1994, 2016-10-17:
I'm not using unity I am using zed camera on Ubuntu and JTK1 ,my goal is to detect face and calibrate it from background to map it to 3D model ...this all I want to do  but I m new in this

adujardin, 2016-10-17:
I meant unity as the Ubuntu desktop environment (https://unity.ubuntu.com/), not Unity the game engine

hala1994, 2016-10-17:
ah okey then I will consider it. 
Thank you

adujardin, 2016-10-17:
If you want to model a face, (something like [this](http://reconstructme.net/) or maybe like [this](https://github.com/HRLTY/DenseFace3D) ?) I don't think rtabmap would give you that kind of results. Like ZEDFu, it's meant to scan a room or large scale areas. You probably have to look for specialized software to get that level of accuracy on small objects.

hala1994, 2016-10-17:
ReconstructMe does it support zed camera ?

adujardin, 2016-10-17:
I don't think so, it looks like it's made for RGB-D sensors like Kinect

hala1994, 2016-10-17:
do you know any other software that can work with zed camera?

adujardin, 2016-10-17:
No sorry, I'm not aware of any software that could correspond to your use case, compatible out of the box with the ZED (It might exist though). The ZED was not really meant for that kind of applications.

Anyway, I'm going to close this issue since it's not related to a ROS wrapper bug.
Good luck !

closed 2016-10-17 by adujardin