stereolabs/zed-ros-wrapper #82: Can I use zed-ros-wrapper without screen?

dgyx, 2016-10-22:
when i run zed on a computer without screen,there is an error:
ZED SDK >> (Init) Best GPU Found : GeForce GTX 750 Ti , ID : 0
ZED SDK >> (Init) Disparity mode has been set to QUALITY
[ INFO] [1477106869.260153825]: ZED_NOT_AVAILABLE

dgyx, 2016-10-22:
My OS is ubuntu 14

nesnes, 2016-10-24:
Hi,
Can you tell me how you are launching the wrapper? Is it with an ssh session?
Can you try to launch the wrapper when connecting with different ssh parameters:
`ssh`, `ssh -X` and `ssh -Y`? Can you give me the result for the three commands?

If you're not starting the node with an ssh session, you may have to deal with `export DISPLAY=:0.0` commands on the target computer.

dgyx, 2016-10-24:
yes, with an ssh session. I tried the three commands,
the result is same error.
ZED SDK >> (Init) Best GPU Found : GeForce GTX 750 Ti , ID : 0
ZED SDK >> (Init) Disparity mode has been set to QUALITY
[ INFO] [1477106869.260153825]: ZED_NOT_AVAILABLE

adujardin, 2016-10-25:
Hi,
In a terminal, could you try this : `lsusb | grep 2b03:f580 | wc -l`
It will give the number of ZED plugged and recognized.

Also, could you launch the Diagnostic tool ? It's available in `/usr/local/zed/tools/ZED\ Diagnostic`

dgyx, 2016-10-26:
thanks for you help:

cd /usr/local/zed/tools
 ./ZED\ Diagnostic 
**********************************************************************
******************** Starting ZED Diagnostic tool ********************
**********************************************************************
 Detected USB3.0 component :  true
 Detected ZED PID/VID :  true
 Detected ZED as Video Device :  true  under /dev/video 0
 Detected ZED with Full Capabilities:  true
 Detected NVIDIA Card :  true  ( 1 )
 Detected Compute Capability :  true  ( 50 )
 Detected Driver version :  true  ( 7050 )
 Detected Required Memory :  true  ( 2044.69 MB)
 lsusb|grep 2b03:f580|wc -l
1

when I attach a screen ,the zed.launch run successfully. So I think there is no connection problem between  zed and computer.
Maybe it's cause by xserver?I tried xvfb,but failed.

adujardin, 2016-10-26:
I managed to reproduce it.
If no user is logged in a graphical session (even with a screen and the xserver launched), the ZED SDK doesn't seem to find the ZED, but the system does (from the `lsusb` output). The ZED SDK samples don't run either. It's is not link to CUDA since the CUDA samples work fine. 

It seems to be a bug in the ZED SDK. For now, I have no other solution than attaching a screen, but we will try to fix this for the upcoming release.

adujardin, 2016-10-26:
I found the problem and committed a workaround in a separate [branch](https://github.com/stereolabs/zed-ros-wrapper/tree/no-screen). 
The next release of the ZED SDK will include the fix.

dgyx, 2016-10-27:
Thx, the commit upstairs seems not fix the bug, so I need to wait for the next release of the ZED SDK, is it?

adujardin, 2016-10-27:
So you still have the issue with the new branch ?
Because it seems that I'm able to reproduce the exact same problem and the new version solve this on every tested machine.

dgyx, 2016-10-27:
I still have the issue, my OS is ubuntu14, zed SDK is 1.1.0( 1.1.1 seems not support ubuntu14? )

ERROR MSG:
galaxyeye@galaxyeye:~/catkin_ws$ roslaunch zed_wrapper zed.launch 
... logging to /home/galaxyeye/.ros/log/840532e0-9c2a-11e6-9800-708bcd9f71f1/roslaunch-galaxyeye-8224.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://galaxyeye:60854/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/frame_rate: 60
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
 * /zed/zed_wrapper_node/right_topic: right/image_rect_...
 * /zed/zed_wrapper_node/sensing_mode: 1

NODES
  /zed/
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[zed/zed_base_link1-1]: started with pid [8242]
process[zed/zed_base_link3-2]: started with pid [8243]
process[zed/zed_wrapper_node-3]: started with pid [8244]
process[zed/zed_state_publisher-4]: started with pid [8252]
[ INFO] [1477565570.184707088]: ZED_WRAPPER Node initialized
[ INFO] [1477565570.609098300]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GeForce GTX 750 Ti , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
[ INFO] [1477565570.994773354]: ZED_NOT_AVAILABLE

galaxyeye@galaxyeye:~/catkin_ws/src/ros/zed-ros-wrapper$ git checkout
M	CMakeLists.txt
M	cfg/Zed.cfg
Your branch is up-to-date with 'origin/no-screen'.

PS:
ZED Diagnostic tool can't detect the zed,maybe ,i have some fault before.

adujardin, 2016-10-27:
The ZED Diagnostic tool seems to behave similarly (same for the ZED SDK and the wrapper on the master branch) :
- no graphical session :
```
 Detected USB3.0 component :  true
 Detected ZED PID/VID :  true
 Detected ZED as Video Device :  false  under /dev/video -1
```
- logged in :
```
 Detected USB3.0 component :  true
 Detected ZED PID/VID :  true
 Detected ZED as Video Device :  true  under /dev/video 0
```
So I don't see any difference between your issue and the one I can get, except I'm on Ubuntu 16.04 with ZED SDK 1.1.1, but I don't think it has an influence here. On the Jetsons, (on Ubuntu 14.04) the fix with Xlib was already integrated into the ZED SDK and worked fine without a screen.

Just to be absolutely sure, could you try a clean compilation like this pls ?
```BASH
catkin_make clean
catkin_make
```

(You're right, 1.1.0 is the last version to support Ubuntu 14.04.)

dgyx, 2016-10-28:
I tried catkin_make clean,even deleted catkin_ws/devel and catkin_ws/build, but no effect.
unfortunately, i have no computer with Ubuntu 16.04. Maybe I should find a spare computer and reinstall the Ubuntu 16.04.

furushchev, 2017-02-18:
@dgyx Hi, could you successfully execute `ZED Explorer` without screen? How about with `sudo`?
If you can execute `ZED Explorer` only with `sudo`, then it will be the problem about privileges.
Could you check the owner and group of `/dev/video*` files by `ls -lFa /dev/video*`?
You will find output like:

```
crwxrwxrwx+ 1 root video 81, 0  2\u6708 15 22:08 /dev/video0
crwxrwxrwx+ 1 root video 81, 1  2\u6708 15 22:08 /dev/video1
```

This means that file descriptor of the ZED is under group `video`.
You can access this file by joining yourself to `video`  group: `sudo adduser $(whoami) video`
In this command, `$(whoami)` is replaced to your current user name.
After that you can execute `ZED Explorer` even without `sudo`, and hopefully your `zed-ros-wrapper` node.
I hope this helps you.

closed 2017-05-03 by dgyx

ilidar, 2017-09-28:
This issue still not fixed. Not working on following system:
```
Ubuntu 16.04
CUDA 8.0
NVIDIA-375
GeForce GTX 1060
```
Any workaround ?