stereolabs/zed-ros-wrapper #83: problem of libsl_tracking.so

honeyasong, 2016-10-25:
hi，

My os is ubuntu 16.04 LTS. I have updated to the cuda-8.0,and I install the ZED sdk of V1.1.1. I met this warning "can't get the status of /usr/local/zed/lib/libsl_tracking.so,because of too many levels of symbolic links ". 

After the installation, I can open the ZED Exploer、ZED Calibration、ZED Diagnostic, but I can't open the  ZED Depth Viewer and ZED SVOEditor. I tried to open the ZED Depth Viewer(I changed the name) in the terminal: ./ZED_Depth_Viewer ,the problem is:"./ZED_Depth_Viewer: error while loading shared libraries: libsl_tracking.so: cannot open shared object file: No such file or directory"

Can anyone help me?Thank you!

adujardin, 2016-10-25:
Hi,
So from your comment in issue #42, removing the folder `/usr/local/zed/` and reinstalling solved your issue ?

honeyasong, 2016-10-25:
@adujardin 
Hi,
Yes,It can now work in my pc.

closed 2016-10-25 by adujardin

honeyasong, 2016-10-25:
Yes. 祝好 李松雨 Best Wishes Songyu Li On 2016-10-25 15:45 , Aymeric Dujardin Wrote: Hi, So from your comment in issue #42, removing the folder /usr/local/zed/ and reinstalling solved your issue ?