stereolabs/zed-ros-wrapper #84: No compatible NVIDIA device found, disparity map unavailable.   Err Code : 8 : 0

shashwat14, 2016-10-26:
I am trying to use the zed-ros-wrapper. My ZED SDK version is 1.1, Linux is 14.04, CUDA is 7.5 and GPU is GTX 1070. Everytime I do a roslaunch zed.launch I see the following error. I am not sure how to fix this. Following shows the error. I had made the change on line 30 regarding CUDA 7.5 instead of CUDA 8.0. Am I missing something here? Thanks.

scot@scot:~$ roslaunch zed_wrapper zed.launch 
... logging to /home/scot/.ros/log/e29328ae-9b48-11e6-941f-80a589653000/roslaunch-scot-12586.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://scot:54578/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/frame_rate: 60
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
 * /zed/zed_wrapper_node/right_topic: right/image_rect_...
 * /zed/zed_wrapper_node/sensing_mode: 1

NODES
  /zed/
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)

auto-starting new master
process[master]: started with pid [12598]
ROS_MASTER_URI=http://scot:11311

setting /run_id to e29328ae-9b48-11e6-941f-80a589653000
process[rosout-1]: started with pid [12611]
started core service [/rosout]
process[zed/zed_base_link1-2]: started with pid [12620]
process[zed/zed_base_link3-3]: started with pid [12629]
process[zed/zed_wrapper_node-4]: started with pid [12636]
process[zed/zed_state_publisher-5]: started with pid [12641]
[ INFO] [1477464803.035674943]: ZED_WRAPPER Node initialized
[ INFO] [1477464803.073979143]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GeForce GTX 1070 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) No compatible NVIDIA device found, disparity map unavailable. 
 Err Code : 8 : 0
[ INFO] [1477464809.764398958]: NO_GPU_COMPATIBLE
ZED SDK >> (Init) Best GPU Found : GeForce GTX 1070 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
 
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) A lot of corrupted frames (frame glitch or invalid size) has been detected. Please check that your ZED is connected to an USB3.0 port, that the USB3.0 chipset driver is up to date...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) No compatible NVIDIA device found, disparity map unavailable. 
 Err Code : 8 : 0
[ INFO] [1477464822.638539857]: NO_GPU_COMPATIBLE

adujardin, 2016-10-26:
Hi,
You have to install CUDA 8.0 for Pascal GPUs like your GTX1070, CUDA 7.5 is not compatible.
Could you launch the diagnostic tool ? It's available in `/usr/local/zed/tools/ZED\ Diagnostic`

tongtybj, 2016-11-23:
@adujardin 
I also have the same problem.
But, I can build and execute the sample form "NVIDIA_CUDA-7.5_Samples" with Uubntu 14.04, GTX 1060, CUDA7.5.

adujardin, 2016-11-23:
Hi,
The CUDA sample from CUDA 7.5 can work because it embeds the PTX instructions which are then recompiled for your card when your run it.

For the ZED SDK, we only provide the final assembly code for each compute capabilities. Therefore we don't provide Pascal compatible code (e.g compute capabilities 6.X) for the installer compiled with CUDA 7.5.

In other words, the ZED SDK which requires CUDA 7.5 is NOT compatible with any Pascal cards (GTX 10XX), no workarounds possible.

tkelestemur, 2016-12-22:
I have the exact setup with @shashwat14 and I encounter with the same problem.
I am able to run ZED Explorer but not ZED Depth Viewer, it gives same error.
So, does that mean we are unable to get depth data with this configuration ?

I have other stuff requires Ubuntu 14.04 and not possible to upgrade 16.04.
I also spent to much time trying to make it run, you should add this issue to website so people will know before starting.

This is my output of ZED Diagnostic:

```

*******************************
*** OS Version ***
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty

Linux river-dicle 4.4.0-57-generic #78~14.04.1-Ubuntu SMP Sat Dec 10 00:14:47 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

*******************************
*** Hardware ***
H/W path         Device     Class          Description
======================================================
                            system         Z170X-UD3 (To be filled by O.E.M.)
/0                          bus            Z170X-UD3-CF
/0/0                        memory         64KiB BIOS
/0/3d                       memory         128KiB L1 cache
/0/3e                       memory         128KiB L1 cache
/0/3f                       memory         1MiB L2 cache
/0/40                       memory         8MiB L3 cache
/0/41                       processor      Intel(R) Core(TM) i7-6700K CPU @ 4.00GHz
/0/42                       memory         32GiB System Memory
/0/42/0                     memory         8GiB DIMM Synchronous 2133 MHz (0.5 ns)
/0/42/1                     memory         8GiB DIMM Synchronous 2133 MHz (0.5 ns)
/0/42/2                     memory         8GiB DIMM Synchronous 2133 MHz (0.5 ns)
/0/42/3                     memory         8GiB DIMM Synchronous 2133 MHz (0.5 ns)
/0/100                      bridge         Sky Lake Host Bridge/DRAM Registers
/0/100/1                    bridge         Sky Lake PCIe Controller (x16)
/0/100/1/0                  display        NVIDIA Corporation
/0/100/1/0.1                multimedia     NVIDIA Corporation
/0/100/14                   bus            Sunrise Point-H USB 3.0 xHCI Controller
/0/100/14.2                 generic        Sunrise Point-H Thermal subsystem
/0/100/16                   communication  Sunrise Point-H CSME HECI #1
/0/100/17                   storage        Sunrise Point-H SATA controller [AHCI mode]
/0/100/1b                   bridge         Sunrise Point-H PCI Root Port #17
/0/100/1b.2                 bridge         Sunrise Point-H PCI Root Port #19
/0/100/1b.2/0    eth1       network        82574L Gigabit Network Connection
/0/100/1c                   bridge         Sunrise Point-H PCI Express Root Port #1
/0/100/1c/0                 bridge         Intel Corporation
/0/100/1c/0/0               bridge         Intel Corporation
/0/100/1c/0/1               bridge         Intel Corporation
/0/100/1c/0/2               bridge         Intel Corporation
/0/100/1c/0/2/0             bus            Intel Corporation
/0/100/1c/0/4               bridge         Intel Corporation
/0/100/1c.4                 bridge         Sunrise Point-H PCI Express Root Port #5
/0/100/1d                   bridge         Sunrise Point-H PCI Express Root Port #9
/0/100/1d.4                 bridge         Sunrise Point-H PCI Express Root Port #13
/0/100/1f                   bridge         Sunrise Point-H LPC Controller
/0/100/1f.2                 memory         Memory controller
/0/100/1f.3                 multimedia     Sunrise Point-H HD Audio
/0/100/1f.4                 bus            Sunrise Point-H SMBus
/0/100/1f.6      eth0       network        Ethernet Connection (2) I219-V
/0/1             scsi0      storage        
/0/1/0.0.0       /dev/sda   disk           512GB Samsung SSD 850
/0/1/0.0.0/1                volume         511MiB Windows FAT volume
/0/1/0.0.0/2     /dev/sda2  volume         444GiB EXT4 volume
/0/1/0.0.0/3     /dev/sda3  volume         31GiB Linux swap volume
/1                          power          To Be Filled By O.E.M.

[... 46 lines cut ...]
>> setupDevice for 1280x720@15 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1280x720@30 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1280x720@60 result : 1  (Note : returned Size was: 1280x720)
>> setupDevice for 1920x1080@15 result : 1  (Note : returned Size was: 1920x1080)
>> setupDevice for 1920x1080@30 result : 1  (Note : returned Size was: 1920x1080)
>> setupDevice for 2208x1242@15 result : 1  (Note : returned Size was: 2208x1242)

 OK : ZED can setup all resolutions

 ZED Serial Number : 2201
 ZED Firmware Number : 1142

*******************************
*** CUDA Capabilities***

 *** CUDA Check *** 
(SMI) : Driver Version                      : 367.57
(SMI) :     Driver Model
(SMI) :     Performance State               : P8
cuDeviceGetCount : OK : found 1 NVIDIA Graphic card. Taking dev 0
cuDeviceGetName : OK : GeForce GTX 1070
cuDeviceComputeCapability :  Compute Cap :  6 1 -- must be between 2.0 and 5.3 
cuDriverGetVersion :  CUDA Driver version : 8000 . Must be >=7500 for CUDA 7.5 toolkit.
cuDeviceTotalMem:   CUDA Memory  4096 MBytes (4294967295 bytes)
```

sl-braun, 2016-12-22:
Hi,

GeForce 10XX cannot be used with CUDA 7.5. They are only supported with CUDA 8.0 (new architecture).
You either need to upgrade to Ubuntu16 and CUDA 8.0 to be able to use the 1070 or use a GeForce 980 (with Ubuntu14 and CUDA 7.5).

Best,
OB

closed 2018-06-28 by Myzhar