stereolabs/zed-ros-wrapper #88: Jetson tk1 , namespace 'sl::zed' does not name a type

Joylim, 2016-11-11:
error:  'TRACKING_STATE' in namespace 'sl::zed' does not name a type

ZED sdk: 0.94
Jetpack: 21.3

adujardin, 2016-11-14:
Hi,

The positional tracking capability was added in the ZED SDK 1.0.0. 
The ROS driver in the master branch is meant for the current ZED SDK version. So you should update the ZED SDK, you can download it [here](https://www.stereolabs.com/developers/).

However, if you want to stay at 0.9.4, you can download [this](https://github.com/stereolabs/zed-ros-wrapper/archive/v0.9.X.zip) old version of ROS wrapper. Please note however that these versions are unsupported, slower and might contain bugs.

Joylim, 2016-11-16:
Thank you very much!!! @adujardin

closed 2016-11-16 by Joylim