stereolabs/zed-ros-wrapper #89: ROS record compressed image error

redheli, 2016-11-19:
Hi,

   I try to create rosbag for compressed image topics(for example /zed/rgb/image_rect_color/compressed),but get errors like below:

> [ERROR] [1479541847.873457515]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)

I am on Ubuntu 16.04 and ros-kinect. 

> max@max-Alienware-15:zed-ros-wrapper(master)$ roslaunch zed_wrapper zed.launch 
> ... logging to /home/max/.ros/log/8b0dc15c-ad27-11e6-981b-1c872cc63ee0/roslaunch-max-Alienware-15-8878.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://max-Alienware-15:46465/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.5
>  * /zed/zed_description: <?xml version="1....
>  * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
>  * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
>  * /zed/zed_wrapper_node/frame_rate: 60
>  * /zed/zed_wrapper_node/gpu_id: -1
>  * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
>  * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
>  * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
>  * /zed/zed_wrapper_node/odometry_db: 
>  * /zed/zed_wrapper_node/odometry_topic: odom
>  * /zed/zed_wrapper_node/openni_depth_mode: 0
>  * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
>  * /zed/zed_wrapper_node/quality: 1
>  * /zed/zed_wrapper_node/resolution: 2
>  * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
>  * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
>  * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
>  * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
>  * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
>  * /zed/zed_wrapper_node/right_topic: right/image_rect_...
>  * /zed/zed_wrapper_node/sensing_mode: 1
>  * /zed/zed_wrapper_node/svo_filepath: 
>  * /zed/zed_wrapper_node/zed_id: 0
> 
> NODES
>   /zed/
>     zed_base_link1 (tf2_ros/static_transform_publisher)
>     zed_base_link3 (tf2_ros/static_transform_publisher)
>     zed_state_publisher (robot_state_publisher/state_publisher)
>     zed_wrapper_node (zed_wrapper/zed_wrapper_node)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[zed/zed_base_link1-1]: started with pid [8896]
> process[zed/zed_base_link3-2]: started with pid [8897]
> process[zed/zed_wrapper_node-3]: started with pid [8903]
> process[zed/zed_state_publisher-4]: started with pid [8913]
> [ INFO] [1479430021.268613951]: Initializing nodelet with 8 worker threads.
> [ INFO] [1479430021.393226265]: Using ZED Camera
> ZED SDK >> (Init) Best GPU Found : GeForce GTX 970M , ID : 0
> ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
> ZED SDK >> (Init) Creating ZED GPU mem...
> ZED SDK >> (Init) Starting Self-Calibration in background... 
> ZED SDK >> (Init) Done...
> [ INFO] [1479430023.824428886]: SUCCESS
> [ INFO] [1479430025.833579572]: Reconfigure confidence : 80
> [ INFO] [1479430026.037812316]: Advertized on topic rgb/image_rect_color
[... 158 lines cut ...]
> [ERROR] [1479432477.172638854]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.262991969]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.348006376]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.423199126]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.532114978]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
> [ERROR] [1479432477.532905228]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
> [ERROR] [1479432477.563674479]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.634002600]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.704204797]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.772292631]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.844905247]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432477.915196282]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.025437855]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
> [ERROR] [1479432478.026099248]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
> [ERROR] [1479432478.054347591]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.123600022]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.192431102]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.262237441]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.331440843]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.401848932]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.505922014]: Compressed Image Transport - JPEG compression requires 8/16-bit color format (input format is: 32FC1)
> [ERROR] [1479432478.506631947]: cv_bridge exception: '[32FC1] is not a color format. but [bgr8] is. The conversion does not make sense'
> [ERROR] [1479432478.539140866]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
> [ERROR] [1479432478.601336152]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
>

kyesh, 2016-12-23:
I get the same error Also running 16.04, Kinetic. I would like to add that as far as I can tell it still records just at like 1 frame/second. 

*Note: I'm playing back via an SVO because my ZED camera has other issues*

danielsnider, 2017-04-21:
Me too, same issue

SebReuter, 2017-05-21:
set openni_depth_mode to 1:
zed_camera.launch:    <param name="openni_depth_mode"     value="1" />

closed 2018-06-28 by Myzhar

bsubbaraman, 2019-11-26:
As a note in case it helps anyone else: using SDK version 2.8.3, in order to record all topics in a rosbag, you must run in `openni_depth_mode` (as noted in this thread), and exclude compressed images by running:
`rosbag record -a -x "(.*)/compressed(.*)"`