stereolabs/zed-ros-wrapper #91: Not enough GPU memory

bnascimento, 2016-11-23:
Guys, we have an erratic problem launching an application on TX1. Sometimes (not every time) we have the message: "ZED SDK >> (Init) Not enough memory for this depth calculation mode..." 
We are launching an application on TX1 running at FULLHD (1080) with 15 fps.
Still didnt found a proper way to remove this error, sometimes a reboot of the dev board helps. We find this particularry weird since we can launch the depth view tool without any problem every time. 
Is there any way we can overcome this problem? Something missing perhaps?
Our initialization code is simple enough and quite similar to the opencv sample here: https://github.com/stereolabs/zed-opencv/blob/master/src/main.cpp

Thank you for your time,
Cheers
Bruno

bnascimento, 2016-11-24:
Like I said before running your tools we never have this error, and running "ZED Depth Viewer" tool we get the following when lauching it from command line:

_ZED SDK >> (Init) Best GPU Found : NVIDIA Tegra X1 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
ZED SDK >> (Init) Done..._

instead of just:

_ZED SDK >> (Init) Not enough memory for this depth calculation mode..._

Which seems like a more proper way to initialize the SDK. What are the steps for the correct initialization? 

Thanks
Cheers guys

bnascimento, 2016-11-24:
Ok, got it, to get that much information, we have to set initialization parameter verbose to true. Still the problem with GPU memory persists. Any additional hints?

nesnes, 2016-11-28:
Hi,
As the error describe, the memory error is linked to the depth mode you're using. You're probably successfully opening the Depth Viewer because its depth mode is set in PERFORMANCE. (it also signify performances in memory usage). Moreover, if the default resolution of the DepthViewer is lower than the resolution used in your program, it can also explain the lower memory requirements.
When you have the issue, try to launch another version of your program that uses PERFORMANCE mode in VGA. 

Best

closed 2018-06-28 by Myzhar