stereolabs/zed-ros-wrapper #93: zed_wrapper zed.launch fails

ThaHypnotoad, 2016-12-12:
when following the instructions [here](http://wiki.ros.org/zed-ros-wrapper) I get this error when testing the launch file:

`[ERROR] [1481518490.283929123]: Failed to load nodelet [/zed/zed_wrapper_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/aleks/Documents/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_calib3d.so.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1481518490.283984374]: The error before refreshing the cache was: Failed to load library /home/aleks/Documents/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_calib3d.so.3.1: cannot open shared object file: No such file or directory`

I've got opencv 3.1 Do I need to give it paths somehow?

ThaHypnotoad, 2016-12-12:
TL;DR: for some reason ros installs opencv rather differently, so I had to build 3.1 from source to get the shared library files where they needed to be.

Interesting. When building the samples I ran into the same issue, and decided to look around for my opencv 3.1 installation. According to python I had it installed, particularly through ros when calling cv2.__file__, however when going there I managed to find only one file related to opencv. Maybe I wasn't looking hard enough but I could not find the rest of the files. I decided to just give up, install opencv 3.1 using their instructions from source, and voila, everything wen't fine.

closed 2016-12-12 by ThaHypnotoad

ChrisPsa, 2017-03-09:
Another workaround, if ROS Kinetic installs the libopencv_*.so.3.2 shared libraries, is to create symbolic links from the files zed_ros_wrapper wants to those files.

`libopencv_*.so.3.1 -> lilbopencv_*.so.3.2`

alexge233, 2017-06-20:
I can confirm this is still an issue present, when using opencv 3.2 and the ros Zed nodelet on Ubuntu 16.04

CleopatraDion, 2017-12-21:
Did anybody ever solve this issue?

agneevguin, 2018-03-14:
Any solutions anyone?

ram651991, 2018-08-24:
I solved the error following [this](https://github.com/tensorflow/tensorflow/issues/19362). But now  found a different error (NO GPU found).
[image]

ram651991, 2018-08-24:
Fixed NO_GPU error by selecting proprietary version- nvidia-384 and restarting the PC.
[image]

itabr, 2018-09-11:
I can confirm this is still an issue present, the above solution did not work for me.

Myzhar, 2018-09-11:
@itabr all the solutions for this kind of problem are available in the previous posts.
There are many causes, you must check which one is generating your error message