stereolabs/zed-ros-wrapper #94: Set Static ID for ZED Camera

reeve333, 2016-12-12:
Hi,

Could I know is that possible to set static ID to the Zed Camera? 

My scenario is I have 4 Zed Camera launching at the same time, but I couldn't stabilize it (Camera switching all the time) and couldn't set fix the camera topic. Could I get suggestion how to fix it or request for the feature?

Meanwhile, I have update some coding for multi-camera launching. Should I make a proper pull request to the git?

Thank you very much.

surirohit, 2017-05-22:
Hi! I have the same issue. The udev attributes are also exactly the same. Did you manage to solve the problem?

tolster710, 2017-12-19:
Same issue here, There are cameras on both sides of the bot, and on boot, it routinely assigns these incorrectly.

adujardin, 2017-12-20:
Hi,

This feature will be included in the update for the upcoming ZED SDK release (2.3). The ZED identification will use its serial number.

tcassell94, 2018-01-19:
For those who need an immediate solution, try this. The requirements are that each ZED camera is plugged into its own USB3.0 port (no USB3.0 hubs allowed).

- Unplug all ZED cameras, except for 1
- See which videoX is left in /dev/videoX
- In a terminal, type:
    `udevadm info -a -n /dev/video0`
- At the parent device that ends in something like "usbX/X-Y", make note of the busnum and devpath attributes.
- Create a file in "/etc/udev/rules.d" called 99-zed.rules.
- Add a line that has the following:
    `ATTRS{idProduct}=="f580", ATTRS{idVendor}=="2b03", ATTRS{busnum}=="X", ATTRS{devpath}=="Y", SYMLINK+="SOME_NAME_HERE"`
- Unplug this current ZED camera, and plug in the other one into its **own usb port**
- Repeat the above for all ZED cameras. 

This basically assigns cameras a /dev name based on which USB port they are plugged into. An ok workaround until serial numbers are added.

commit 3444b37b (2018-01-19): Fix #94 set satic ZED id (via its SN)
closed 2018-01-19 by adujardin