stereolabs/zed-ros-wrapper #98: ZED ros wrapper not working in Jetson TX1

udayakumar97, 2016-12-20:
I am trying to launch zed.launch on Jetson TX1 and I get this error
process[master]: started with pid [9038]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a90b6f1e-c6d2-11e6-8f25-00044b633dea
process[rosout-1]: started with pid [9051]
started core service [/rosout]
process[zed/zed_base_link1-2]: started with pid [9068]
process[zed/zed_base_link3-3]: started with pid [9069]
process[zed/zed_wrapper_node-4]: started with pid [9077]
process[zed/zed_state_publisher-5]: started with pid [9089]
[ INFO] [1482251877.700664233]: Initializing nodelet with 4 worker threads.
[ INFO] [1482251877.968502411]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : NVIDIA Tegra X1 , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
[zed/zed_wrapper_node-4] process has died [pid 9077, exit code -11, cmd /home/ubuntu/zed_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/a90b6f1e-c6d2-11e6-8f25-00044b633dea/zed-zed_wrapper_node-4.log].
log file: /home/ubuntu/.ros/log/a90b6f1e-c6d2-11e6-8f25-00044b633dea/zed-zed_wrapper_node-4*.log

It was compiled successfully but gives this issue when running the launch file. On a little bit debugging I found that as soon as the load method is called by zed_wrapper_node, it crashes. I tried adding a NODELET_INFO_STREAM in the first line of onInit function in the nodelet but nothing was displayed, so I presume that the nodelet is not spawning properly

tanmayag98, 2016-12-21:
Hi,
I got the same error and realized that there are two reasons for this:
1.The camera id in the launch file is not the correct one...it might be taking your webcam as  default.
2.You might be using opencv 3.1 with Jetson TX1.You will have to uninstall that and install the opencv4tegra version.
Hope it helps!!

udayakumar97, 2016-12-21:
Thanks it worked now

ernestcheung, 2017-02-07:
Hi, I am having the same issue.  
But I have already set:
1) zed_id:=1 when starting the launch file
2) Set OpenCV_INCLUDE_DIRS and OpenCV_LIBRARY_DIRS to the opencv4tegra version to avoid it pointing to the ros kinetic default version.
But I still got the same error.  

Am I doing it wrongly?
Can @udayakumar97 share your way of doing it?

udayakumar97, 2017-02-08:
Did you set CUDA_VERSION 8.0 and CUDA_STATIC_RUNTIME OFF?
Make sure you are not explicitly setting any path for python

ElimsV, 2017-07-03:
I have similar problem with Jetson TK1 and haven't solved it yet.
The error message:
---------------------------------------------------------------------------------------------------------------------------------------
... logging to /home/ubuntu/.ros/log/fb2b5c5c-1dd6-11b2-8e95-00044b49198a/roslaunch-tegra-ubuntu-3520.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.2:36806/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/frame_rate: 60
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_raw_topic: right/image_raw_c...
 * /zed/zed_wrapper_node/right_topic: right/image_rect_...
 * /zed/zed_wrapper_node/sensing_mode: 1
 * /zed/zed_wrapper_node/svo_filepath: 
 * /zed/zed_wrapper_node/zed_id: 0

NODES
  /zed/
    zed_base_link1 (tf2_ros/static_transform_publisher)
    zed_base_link3 (tf2_ros/static_transform_publisher)
    zed_state_publisher (robot_state_publisher/state_publisher)
    zed_wrapper_node (zed_wrapper/zed_wrapper_node)

ROS_MASTER_URI=http://192.168.2.2:11311

core service [/rosout] found
process[zed/zed_base_link1-1]: started with pid [3538]
process[zed/zed_base_link3-2]: started with pid [3539]
process[zed/zed_wrapper_node-3]: started with pid [3540]
process[zed/zed_state_publisher-4]: started with pid [3541]
[ INFO] [4293.888484954]: Initializing nodelet with 4 worker threads.
[ INFO] [4295.072637485]: Using ZED Camera
ZED SDK >> (Init) Best GPU Found : GK20A , ID : 0
ZED SDK >> (Init) Disparity mode has been set to PERFORMANCE
ZED SDK >> (Init) Creating ZED GPU mem...
ZED SDK >> (Init) Starting Self-Calibration in background... 
[zed/zed_wrapper_node-3] process has died [pid 3540, exit code -9, cmd /home/ubuntu/el2425_ws/devel/lib/zed_wrapper/zed_wrapper_node __name:=zed_wrapper_node __log:=/home/ubuntu/.ros/log/fb2b5c5c-1dd6-11b2-8e95-00044b49198a/zed-zed_wrapper_node-3.log].
log file: /home/ubuntu/.ros/log/fb2b5c5c-1dd6-11b2-8e95-00044b49198a/zed-zed_wrapper_node-3*.log
---------------------------------------------------------------------------------------------------------------------------------------
The exit code is -9, indicating process gets killed.
ZED Explorer could be ran successfully. However running samples via terminal will return message "Killed".
I am using CUDA 6.5, ZED SDK 1.2 (required by Jetson TK1) and opencv4tegra (basing on opencv 2.4.12). Also, I made sure ROS find the right version of opencv(2.4.12) according to
http://myzharbot.robot-home.it/blog/software/ros-nvidia-jetson-tx1-jetson-tk1-opencv-ultimate-guide/
Not sure how to handle this problem. Appreciate any kind of suggestion.

Best,

ElimsV, 2017-07-04:
Update: changing USB2.0 port to 3.0 solves my problem

closed 2018-06-28 by Myzhar