stereolabs/zed-ros-wrapper #100: Failed to load nodelet [/zed/zed_wrapper_node]

ameerhamzakhan0310, 2017-01-13:
After installing zed-ros-wrapper, I launched 'zed.launch' but it gave the following error:

> [ERROR] [1484294711.176260141]: Failed to load nodelet [/zed/zed_wrapper_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/ameer/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_calib3d.so.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1484294711.176341483]: The error before refreshing the cache was: Failed to load library /home/ameer/catkin_ws/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_calib3d.so.3.1: cannot open shared object file: No such file or directory)

How to fix this error?

ameerhamzakhan0310, 2017-01-13:
The solution can be found [here](https://github.com/stereolabs/zed-ros-wrapper/issues/93).

closed 2017-01-13 by ameerhamzakhan0310

dineshrboson, 2021-02-03:
### Zed launch error

[ WARN] [1612354078.422513277]: The 'state_publisher' executable is deprecated. Please use 'robot_state_publisher' instead
[ INFO] [1612354078.439180063]: Initializing nodelet with 4 worker threads.
[ERROR] [1612354078.474783373]: Failed to load nodelet [/zed/zed_wrapper_node] of type [zed_wrapper/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/dinesh/self-driving-golf-cart/ros/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnppial.so.10: cannot open shared object file: No such file or directory)
[ERROR] [1612354078.474844875]: The error before refreshing the cache was: Failed to load library /home/dinesh/self-driving-golf-cart/ros/devel/lib//libZEDWrapper.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libnppial.so.10: cannot open shared object file: No such file or directory)

Myzhar, 2021-02-03:
`Poco exception = libnppial.so.10` there is something wrong with your CUDA installation

dineshrboson, 2021-02-03:
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Nov__3_21:07:56_CDT_2017
Cuda compilation tools, release 9.1, V9.1.85

dineshrboson, 2021-02-03:
Hey thanks for the reply very humbled ........I Installed zed sdk 2.8 version with cuda 10.2 but its showing.......cuda v9.(I had both cuda 9 10 11 in mu usr/local)......but I got my compilation done and getting launch file error in zed.....

Myzhar, 2021-02-03:
If you change the CUDA version between different compilations you must delete the folders `devel` and `build` in the root of your catkin workspace before each new compilation to clean the CMake cache