stereolabs/zed-ros-wrapper #105: Jetson TX1 - error during ROS wrapper compilation

Ashram56, 2017-02-13:
Good morning,

I have been trying for quite some time to get the ZED ROS wrapper working, but no luck so far.

I have installed PCL from source, as well as VTK
I installed ROS Kinetic using this link : https://github.com/jetsonhacks/installROSTX1

But when running catkin_make, I get the following error during link:

/usr/bin/ld: cannot find -lvtkCommon
/usr/bin/ld: cannot find -lvtkFiltering
/usr/bin/ld: cannot find -lvtkImaging
/usr/bin/ld: cannot find -lvtkGraphics
/usr/bin/ld: cannot find -lvtkGenericFiltering
/usr/bin/ld: cannot find -lvtkIO
/usr/bin/ld: cannot find -lvtkRendering
/usr/bin/ld: cannot find -lvtkVolumeRendering
/usr/bin/ld: cannot find -lvtkHybrid
/usr/bin/ld: cannot find -lvtkWidgets
/usr/bin/ld: cannot find -lvtkParallel
/usr/bin/ld: cannot find -lvtkInfovis
/usr/bin/ld: cannot find -lvtkGeovis
/usr/bin/ld: cannot find -lvtkViews
/usr/bin/ld: cannot find -lvtkCharts
collect2: error: ld returned 1 exit status
zed-ros-wrapper/CMakeFiles/ZEDWrapper.dir/build.make:221: recipe for target '/home/ubuntu/catkin_ws/devel/lib/libZEDWrapper.so' failed
make[2]: *** [/home/ubuntu/catkin_ws/devel/lib/libZEDWrapper.so] Error 1
CMakeFiles/Makefile2:4109: recipe for target 'zed-ros-wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Since I compiled completely vtk from source, I would have thought these libraries would have been installed.

I can't figure out how to install these missing libraries, any pointers ?

Thanks

Vincent

griz1112, 2017-02-15:
python-vtk It will remove libvtk so no need to uninstall. Without this only raw image transport is avilable.

ShanLuo, 2017-02-17:
Hi @griz1112 , thanks for your instruction and I have installed vtk6 by 'sudo apt-get install python-vtk6
', but I still got the error as @Ashram56 had. Any advice? Thanks!

Hi @Ashram56 , have you solved this issue?

closed 2018-06-28 by Myzhar